ophub/amlogic-s9xxx-openwrt #555: 软件源更新失败，写入emmc后无法启动。

lccxsusuk, 2024-03-15:
设备信息：
Board: s905l3b | OpenWrt Kernel: 6.6.21-ophub
 Builder Name: ophub | Production Date: 2024-03-12
───────────────────────────────────────────────────────────────────────
 Device Model: Amlogic Meson GXL (S905L3b) MBH-M302A Box
 Architecture: AArch64 : Cortex-A53 x 4

固件版本
OpenWrt_lede_save_2024.03  

纯小白请教，共有个两问题，
1 更新失败，opkg update 提示
Collected errors:
 * opkg_download: Failed to download https://mirrors.cloud.tencent.com/lede/snapshots/targets/armvirt/64/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

2写入emmc后启动失败，查看资料，需给Bl打勾，web端没有BL 勾选选项。
[image]

ophub, 2024-03-15:
插件设置里有选项

lccxsusuk, 2024-03-15:
大佬  我换用[OpenWrt_immortalwrt_save_2024.03](https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_immortalwrt_save_2024.03)，依然不能更新，请教一下，怎样设置才能更新。

lccxsusuk, 2024-03-15:
大佬  我换用[OpenWrt_immortalwrt_save_2024.03](https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_immortalwrt_save_2024.03)，依然不能更新，请教一下，怎样设置才能更新。
[image]

closed 2024-03-15 by lccxsusuk

ophub, 2024-03-15:
你的网络ping不通这个源吗？

lccxsusuk, 2024-03-15:
大佬 更新源的地址在浏览器是打不开的。https://mirrors.vsean.net/openwrt/snapshots/targets/armsr/armv8/packages.