ophub/amlogic-s9xxx-openwrt #556: 大佬您好，请问有支持S905-B的openwrt系统吗？

nihaonas, 2024-03-17:
大麦盒子 DM4036  用的S905-B的芯片 请问这个适合您的哪个包呢？或者可否添加支持呢？谢谢！

ophub, 2024-03-17:
你试试s905的

nihaonas, 2024-03-17:
感谢大佬回复，
[image]
请问
1.这个后缀名gd的 和sha的 用哪个呢？ 
2.还有k5 / k6是什么意思用哪个呢？
3.还有那个dtb改用哪个呢？
希望得到大佬回复，感谢感谢！

ophub, 2024-03-17:
https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/armbian-files/common-files/etc/model_database.conf

参考相似的改
5.15.y这个是内核的版本

nihaonas, 2024-03-17:
[image]
我用了P200的，我看恢复模式下提示 是P200，但是用开心电视助手连接abd，点击从U盘启动后还是这样进入恢复模式界面，这是没有引导成功是吗？
我拆机了 以下是主板的图片
[image]
[image]
[image]

nihaonas, 2024-03-17:
这个是主板的图片
[image]
[image]
[image]

nihaonas, 2024-03-17:
> https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/armbian-files/common-files/etc/model_database.conf
> 
> 参考相似的改 5.15.y这是内核的版本

不好意思大佬，我是个半桶水，看不懂这个 ，大佬能帮忙搞一个可以用的吗？谢谢

nihaonas, 2024-03-18:
meson-gxbb-p201.dtb  
meson-gxbb-mxq-pro-plus.dtb  
meson-gxbb-beelink-mini-mx.dtb  
三个dtp我都试用了，都不能从U盘开机，不知道是我的盒子不支持从U盘开机，还是文件内核不对

ophub, 2024-03-18:
看文档，如何从安卓其他USB，你开启调试模式。

closed 2024-03-18 by ophub

ldw2545398, 2025-04-08:
解决了么？ 同求~

nihaonas, 2025-05-25:
> 同求~

没有解决  你的解决了吗？

billdegithub, 2025-06-05:
研究了，根本不行。s905-b是高安芯片。u-boot就失败了。。ttl强制u-boot进入，手工加载
usb start
fatload usb 0:1 0x1000000 uImage
fatload usb 0:1 0x2000000 uInitrd
fatload usb 0:1 0x3000000 dtb/amlogic/meson-gxbb-p200.dtb

setenv bootargs console=ttyS0,115200 root=/dev/sda1 rootwait
bootm 0x1000000 0x2000000:${filesize} 0x3000000
提示：签名问题。
aml log : Sig Check 967

把ulmage改为这个机顶盒的自带boot.img 导入测试，又有后面的加密规则提示signature: erase_lablbootm 0x1000000
fdt_instaboot: get header err。总之实现不了。。这鸡巴高安芯片。