ophub/amlogic-s9xxx-openwrt #563: Please Add Device Nanopi R2S

adibindrianto, 2024-04-04:
please add support for nanopi r2s devices, thank you

reukiodo, 2024-04-09:
Isn't it already supported officially? https://openwrt.org/toh/friendlyarm/nanopi_r2s

ophub, 2024-04-09:
Added, could you please test if it works as expected?

adibindrianto, 2024-04-09:
I will try and report any bugs found

adibindrianto, 2024-04-09:
1. Can't access the web UI
2. The indicator light on the device does not turn on
3. Changing to another kernel still returns to kernel 6.1

ophub, 2024-04-10:
I am using the default configuration of armbian official r2s, which should work fine. Could you please send me a download link to an armbian image file that you can use normally, so I can take a look and see what the problem is?

adibindrianto, 2024-04-10:
I tried openwrt 23.05.3 kernel ophub stable all versions with build imager

closed 2024-04-27 by ophub