ophub/amlogic-s9xxx-openwrt #570: Install Wireguard

Adam1986, 2024-04-12:
**Device Information | 设备信息**
 - SOC: S905X3
 - Model X96 AIR

**OpenWrt Version | 系统版本**
 - Kernel Version: 6.1.84
 - Release: official - openwrt_official_amlogic_s905x3_k6.1.84_2024.04.08.img.gz

Have just discovered this wonderful project, used the above image on my X96 Air but I am struggling to install wireguard:

```
root@OpenWrt:~# opkg install wireguard-tools
Installing wireguard-tools (1.0.20210914-r3) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_generic/base/wireguard-tools_1.0.20210914-r3_aarch64_generic.ipk
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.82~4f759fb637b59dfd021404e96ce9f182-r1) for kmod-crypto-lib-chacha20
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-lib-chacha20 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.82~4f759fb637b59dfd021404e96ce9f182-r1) for kmod-crypto-lib-poly1305
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-lib-poly1305 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.82~4f759fb637b59dfd021404e96ce9f182-r1) for kmod-crypto-lib-chacha20poly1305
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-lib-chacha20poly1305 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.82~4f759fb637b59dfd021404e96ce9f182-r1) for kmod-crypto-kpp
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-kpp found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.82~4f759fb637b59dfd021404e96ce9f182-r1) for kmod-crypto-lib-curve25519
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-crypto-lib-curve25519 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.82~4f759fb637b59dfd021404e96ce9f182-r1) for kmod-udptunnel4
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-udptunnel4 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.82~4f759fb637b59dfd021404e96ce9f182-r1) for kmod-udptunnel6
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-udptunnel6 found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.82~4f759fb637b59dfd021404e96ce9f182-r1) for kmod-wireguard
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-wireguard found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for wireguard-tools:
 *      kernel (= 6.1.82~4f759fb637b59dfd021404e96ce9f182-r1)
 * opkg_install_cmd: Cannot install package wireguard-tools.
```

Is it safe to install using --force-depends?

Thanks

Adam1986, 2024-04-17:
No feedback?

DynamicLoader, 2024-04-18:
The same, caused by the snapshot kernel used. Currently the kmods are not provided by official, so the alternative way is to re-compile whole image, either full build or just use the image builder. It's suggested to use 5.15.x kernel which has been tested today to work with official opkg source (in my case). 

 :-D

Adam1986, 2024-04-24:
> The same, caused by the snapshot kernel used. Currently the kmods are not provided by official, so the alternative way is to re-compile whole image, either full build or just use the image builder. [...]

How do you go about doing this then? thanks.

DynamicLoader, 2024-04-25:
**re-compile whole image, either full build or just use the image builder**

The full instruction is just in the document, so please refer to it. 

**Or, you can jsut downgrade it back to a version that uses 5.15.x kernel, which feeds are maintained by official and compatible with the kernel here.**

Adam1986, 2024-04-26:
I did try release named "openwrt_official_amlogic_s905x3_k5.15.153_2024.04.08.img.gz" but this was just coming up with a similar error when trying to install wireguard?

closed 2024-04-27 by ophub