ophub/amlogic-s9xxx-openwrt #572: 希望能够支持 海信 hs530r

hunter24099, 2024-04-17:
**Device Information | 设备信息**
 - SOC: [rk3399]
 - Model [海信 hs530r]
 - 4g 运存 
 - 16Gemmc
 - fn-link 6222d-ubb wifi
 - rtl8211
 - usb 2.0 *2
 - usb 3.0 *2
 - hdmi *1

uboot、dtb、以及好用的Armbian_23.08.0-trunk_Hisense-hs530r_jammy_edge_6.5.5_xfce_desktop
链接：https://pan.baidu.com/s/1MNEwM_sln754olAontfrYQ 
提取码：xfjd

感谢老大！

ophub, 2024-04-27:
已经添加

closed 2024-04-27 by ophub

akwangl, 2024-04-30:
这板子有安卓tv吗

hunter24099, 2024-04-30:
安卓据说有，我没找到。我刷了op大佬的op，网络显示百兆，usb3.0好像也驱动起来

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2024年04月30日 09:47 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [ophub/amlogic-s9xxx-openwrt] 希望能够支持 海信 hs530r  (Issue #572) |

这板子有安卓tv吗

yw1818, 2024-05-10:
请问一下，这块板子刷机短接点是哪个点？麻烦指点一下，谢谢

wlau29, 2024-05-13:
> 请问一下，这块板子刷机短接点是哪个点？麻烦指点一下，谢谢

[image]

wlau29, 2024-05-13:
请问一下，这块板子的TTL在哪？

wlau29, 2024-05-14:
这块板的千兆网卡有问题。
```
perf3 -c 192.168.198.202   
Connecting to host 192.168.198.202, port 5201
[  5] local 192.168.198.103 port 44412 connected to 192.168.198.202 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   100 MBytes   841 Mbits/sec    0   3.01 MBytes       
[  5]   1.00-2.00   sec  96.2 MBytes   809 Mbits/sec    0   3.01 MBytes  
```   
------------------
```
iperf3 -c 192.168.198.202 -R
Connecting to host 192.168.198.202, port 5201
Reverse mode, remote host 192.168.198.202 is sending
[  5] local 192.168.198.103 port 44408 connected to 192.168.198.202 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.01   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   1.01-2.01   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   2.01-3.01   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   3.01-4.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   5.00-6.01   sec  0.00 Bytes  0.00 bits/sec  
```        

------------------
用了 "ethtool -K eth0 tx off rx off" 没有帮助。

用了 "ethtool -s eth0 advertise 0xf" 设置成 100mbps, 就可以：
```
iperf3 -c 192.168.198.202 -R
Connecting to host 192.168.198.202, port 5201
Reverse mode, remote host 192.168.198.202 is sending
[  5] local 192.168.198.103 port 44444 connected to 192.168.198.202 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  11.2 MBytes  94.0 Mbits/sec                  
[  5]   1.00-2.00   sec  11.2 MBytes  94.1 Mbits/sec                  
[  5]   2.00-3.00   sec  11.2 MBytes  93.7 Mbits/sec   
```

wlau29, 2024-05-14:
> 这块板的千兆网卡有问题。
>
> ```
> perf3 -c 192.168.198.202
> Connecting to host 192.168.198.202, port 5201
> [  5] local 192.168.198.103 port 44412 connected to 192.168.198.202 port 5201
> [ ID] Interval [...]

用的是：Armbian_24.5.0_rockchip_hs530r_noble_6.1.87_server_2024.05.03.img.gz

wlau29, 2024-05-14:
> 请问一下，这块板子刷机短接点是哪个点？麻烦指点一下，谢谢

短接到地：
[image]

wlau29, 2024-05-14:
> 短接到地： [...]

在板子的背面，emmc的正背面。

yw1818, 2024-05-15:
大佬们，这款板子EMMC可以换成256G吗？是不是直接买颗粒换上

c-hunter007, 2024-05-28:
麻烦哪位大哥分享一下这块板子刷armbian的教程吧？新手刚开始折腾arm板，多谢！

idbloader.img和u-boot.itb是干什么用的？需要一起刷入吗？

xdien, 2024-12-04:
> **Device Information | 设备信息**
> 
>     * SOC: [rk3399]
> 
>     * Model [海信 hs530r]
> 
>     * 4g 运存
> 
>     * 16Gemmc
> 
>     * fn-link 6222d-ubb wifi
> 
>     * rtl8211
> 
>     * usb 2.0 *2
> 
>     * usb 3.0 *2
> 
>     * hdmi *1
> 
> 
> uboot、dtb、以及好用的Armbian_23.08.0-trunk_Hisense-hs530r_jammy_edge_6.5.5_xfce_desktop 链接：https://pan.baidu.com/s/1MNEwM_sln754olAontfrYQ 提取码：xfjd
> 
> 感谢老大！ ![1](https://private-user-images.githubusercontent.com/76616820/323199259-4d4eba79-9fc7-4070-8689-91db0ba38ac2.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzMzMTU4MzksIm5iZiI6MTczMzMxNTUzOSwicGF0aCI6Ii83NjYxNjgyMC8zMjMxOTkyNTktNGQ0ZWJhNzktOWZjNy00MDcwLTg2ODktOTFkYjBiYTM4YWMyLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjA0VDEyMzIxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc0YmRmZTJiZTg5MTE2ZmM2ZjljMGVmMjI4MmUyYzEyNTFhODg3N2JkYjc3MGY1MDY4MmRhMzkxMjRmZDZmNTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.NUgYoknQ3_g6If_p6RVEbCknDDpyapnotukfD_8fh-s)

Can you upload it to another server? I don't have an account to download it, nor can I register because I don't have a mainland China phone number

hunter24099, 2024-12-04:
https://files.kos.org.cn/rockchip/Hisense_hs530r/

---- 回复的原邮件 ----
| 发件人 | Bùi Xuân ***@***.***> |
| 日期 | 2024年12月04日 20:48 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [ophub/amlogic-s9xxx-openwrt] 希望能够支持 海信 hs530r  (Issue #572) |

Device Information | 设备信息

* SOC: [rk3399]

* Model [海信 hs530r]

* 4g 运存

* 16Gemmc

* fn-link 6222d-ubb wifi

* rtl8211

* usb 2.0 *2

* usb 3.0 *2

* hdmi *1

uboot、dtb、以及好用的Armbian_23.08.0-trunk_Hisense-hs530r_jammy_edge_6.5.5_xfce_desktop 链接：https://pan.baidu.com/s/1MNEwM_sln754olAontfrYQ 提取码：xfjd

感谢老大！

Can you upload it to another server? I don't have an account to download it, nor can I register because I don't have a mainland China phone number