ophub/amlogic-s9xxx-openwrt #574: 请求支持设备 DoorNet2

niniwith4cats, 2024-04-18:
ophub您好！在n1上使用您的armbian和openwrt固件很久了，不知道能否为这款设备提供支持？很想使用immortalwrt源码的openwrt。

DHDAXCW的openwrt项目：
https://github.com/DHDAXCW/lede-rockchip

官方设备资料（含debian和ubutu镜像）
https://pan.baidu.com/s/1ppOEYLyaj8gNvQfiwg4Kyw 提取码：6oq6

处理器：Rockchip RK3399

内存：板载双通道DDR3 RAM 1GB

存储：板载EMMC 8GB

双千兆网口：

10/100/1000M自适应以太网口

10/100/1000M自适应PCIE转以太网口

无线网卡：USB2.0无线WIFI网卡，支持IEEE 802.11a/b/g/n/ac，无线连接速率最高可达433Mbps(选配)。

TF卡座：使用microSD卡

USB3.0：USB Type-A接口 x 2

电源接口：USB Type-C接口，5V3A

ophub, 2024-04-18:
根据你提供的资料，我使用了这个网站的u-boot和dts文件，经测试都编译正常，使用的6.1.y的内核测试的，你到releases里下载固件试试。

```
# 仓库：
https://github.com/DHDAXCW/lede-rockchip

# 仓库里的源文件：
./target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3399-doornet2.dts
./target/linux/rockchip/files/arch/arm64/boot/dts/rockchip/rk3399-doornet2.dtsi
./package/boot/uboot-rockchip/src/configs/doornet2-rk3399_defconfig

# 编译完的二进制文件：
./lede-rockchip/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.85/arch/arm64/boot/dts/rockchip/rk3399-doornet2.dtb
./lede-rockchip/build_dir/target-aarch64_generic_musl/u-boot-doornet2-rk3399/u-boot-2024.01/{u-boot.itb,idbloader.img}
```

niniwith4cats, 2024-04-18:
你好，尝试了openwrt和armbian镜像，机器电源红灯常亮，应该是没有正常启动。

ophub, 2024-04-18:
https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_official_save_2024.04/openwrt-rockchip-armv8-embedfire_doornet2-ext4-sysupgrade.img.gz

你试试这个，我用上面的源码本地编译的openwrt默认固件，我手动上传到这的，你测试下。

niniwith4cats, 2024-04-18:
> https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_official_save_2024.04/openwrt-rockchip-armv8-embedfire_doornet2-ext4-sysupgrade.img.gz
> 
> 你试试这个，我用上面的源码本地编译的openwrt默认固件，我手动上传到这的，你测试下。
> 
> 这个设备是不是好几个版本，有1G和4G版？我看lede有2个不同的dtb

你好，链接内的固件可以顺利启动，有线正常，指示灯正常，不过机器自带的无线网卡没有被驱动。

我只见过1G版的，搜了搜别人的评测以及闲鱼之类的平台，也只有1G版。

ophub, 2024-04-19:
我换了个dtb，你再试试

niniwith4cats, 2024-04-19:
> 我换了个dtb，你再试试
> 
> 我去lede仓库看了下，网卡有8168/8169不同的，内存有DDR3 1GB和lpddr4 4gb的，这个产品因为各种乱，已经被lede删除支持了。 https://github.com/search?q=repo%3Acoolsnowwolf%2Flede+DoorNet2&type=pullrequests

我试了下新release内的immortal，不行，无法启动，我已经尝试日用您本地编译那个固件了，还挺好的。

此外NanoPi的R4S倒是1GB DDR3/4GB LPDDR4混用，DN2这个设备我倒是真没见过4GB版...

closed 2024-04-27 by ophub