ophub/amlogic-s9xxx-openwrt #577: S905l2使用方法以及排错经验

liuyi265, 2024-04-24:
**Device Information | 设备信息**
 - SOC: s905l2

刷入img到U盘，从U盘启动，如果无法识别网卡，请复制u-boot-s905x-s912.bin 为 u-boot.ext 如果无法启动，请将extlinux/extlinux.conf.bak  去掉.bak

到此U盘启动已经没问题了，写入emmc,如果未能识别emmc 请下载下面附件的dbt文件，来源于https://github.com/ophub/amlogic-s9xxx-armbian/issues/1278这个地方
[dtb-decompile.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/15090627/dtb-decompile.zip)
替换掉目录中的dbt即可识别，然后写入emmc

写入完成后，我的盒子必须将extlinux.conf 改为extlinux.conf.bak  才能正常启动，但是无法驱动网卡，添加u-boot.emmc后无法开机

到此为止是我折腾这个盒子的全部经验了，留给后面折腾的人吧  如果最后解决了写入emmc网卡驱动的问题希望回复下

ssssshql, 2024-05-10:
目前我也是刷入emmc后没有网络。怎么添加u-boot.emmc，是直接根目录复制一份改名就可以了吗？我这样复制了一份再刷入也还是不行

liuyi265, 2024-05-10:
> 目前我也是刷入emmc后没有网络。怎么添加u-boot.emmc，是直接根目录复制一份改名就可以了吗？我这样复制了一份再刷入也还是不行

教程是这样说的 复制ext改名emmc就行  我也卡到这一步了  我加了这个emmc无法启动  你不需要重新刷机啊 直接操作emmc第一个分区里面的文件就行了

ssssshql, 2024-05-10:
> > 目前我也是刷入emmc后没有网络。怎么添加u-boot.emmc，是直接根目录复制一份改名就可以了吗？我这样复制了一份再刷入也还是不行
> 
> 教程是这样说的 复制ext改名emmc就行 我也卡到这一步了 我加了这个emmc无法启动 你不需要重新刷机啊 直接操作emmc第一个分区里面的文件就行了

直接操作emmc第一个分区的文件是指刷入emmc后再去复制文件嘛

liuyi265, 2024-05-10:
> > > 目前我也是刷入emmc后没有网络。怎么添加u-boot.emmc，是直接根目录复制一份改名就可以了吗？我这样复制了一份再刷入也还是不行
> > 
> > 
> > 教程是这样说的 复制ext改名emmc就行 我也卡到这一步了 我加了这个emmc无法启动 你不需要重新刷机啊 直接操作emmc第一个分区里面的文件就行了
> 
> 直接操作emmc第一个分区的文件是指刷入emmc后再去复制文件嘛

是的

ssssshql, 2024-05-10:
> > > > 目前我也是刷入emmc后没有网络。怎么添加u-boot.emmc，是直接根目录复制一份改名就可以了吗？我这样复制了一份再刷入也还是不行
> > > 
> > > 
> > > 教程是这样说的 复制ext改名emmc就行 我也卡到这一步了 我加了这个emmc无法启动 你不需要重新刷机啊 直接操作emmc第一个分区里面的文件就行了
> > 
> > 
> > 我是直接复制u-boot-s905x-s912.bin这个文件，改成u-boot.ext和u-boot.emmc放到第一个分区，然后extlinux.conf.bak 改为extlinux.conf刷进去的。刷进去之后一直在loading，ssh也连不上。u盘启动的模式是能连ssh的
> > 我也卡在这里了 我看到有个教程说替换个内核就可以  但是我没搞定https://bbs.hassbian.com/thread-17485-1-1.html

谢谢兄弟，我去看看

ssssshql, 2024-05-11:
老哥我这边成功了
[image]
不用替换内核啥的，就是要改一下频率，我改成1000....就ok了。刷入emmc，有线网卡也没问题（无线没试）。
步骤
1. extlinux.conf.bak 改为extlinux.conf
2. u盘的boot分区下,复制u-boot-s905x-s912.bin这个文件，改成u-boot.ext
3. 替换自己改频率的dst
目前在Armbian_24.5.0_amlogic_s905l2_noble_6.1.87_server_2024.05.02.img.gz上测试没有问题了
之前那个帖子都有提到，昨天没认真看，也要看看是不是emmc的芯片

liuyi265, 2024-05-11:
> 老哥我这边成功了 ![Clip_2024-05-11_13-27-13](https://private-user-images.githubusercontent.com/53895670/329756197-f702d189-8f29-4ed0-b57a-e43dd93085a9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU0MDc2MTQsIm5iZiI6MTcxNTQwNzMxNCwicGF0aCI6Ii81Mzg5NTY3MC8zMjk3NTYxOTctZjcwMmQxODktOGYyOS00ZWQwLWI1N2EtZTQzZGQ5MzA4NWE5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTExVDA2MDE1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM0MTNlMDM5MTUxZDNjYzJmZTI4ZjkzODcwYzMzMWE0ZjQ1Y2U1MDE4OWE2MzY2ZjIxNzBkMDgzZDc4YzUzMTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.D5pWdU5ycxhT9S9LZao88GoyPu50WyOLrluHZIh5_xc) 不用替换内核啥的，就是要改一下频率，我改成1000....就ok了。刷入emmc，有线网卡也没问题（无线没试）。 步骤
> 
> 1. extlinux.conf.bak 改为extlinux.conf
> 2. u盘的boot分区下,复制u-boot-s905x-s912.bin这个文件，改成u-boot.ext
> 3. 替换自己改频率的dst
> 
> 这个是我修改后编译的dst： [meson-gxl-s905l2-x7-5g.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/15281730/meson-gxl-s905l2-x7-5g.zip)
> 
> 目前在Armbian_24.5.0_amlogic_s905l2_noble_6.1.87_server_2024.05.02.img.gz上测试没有问题了
> 
> github那个帖子都有提到，昨天没认真看，也要看看是不是emmc的芯片
> 
> 之前我一直不行估计是刷入emmc的时候频率太高导致断断续续的出现很多报错

不需要加u-boot.emmc哇

ssssshql, 2024-05-11:
我这不用加也ok，ssh也连的上，重启了四五次也没问题

liuyi265, 2024-05-11:
> 我这不用加也ok，ssh也连的上，重启了四五次也没问题

我这边不加u-boot.emmc 无法开机
如果把extlinux.conf 删了可以开机，但是网卡不识别  看来我们的问题还不一样

U盘启动加了extlinux.conf  和 u-boot.ext 就一切正常 看来我们的问题还不一样 恭喜恭喜

ssssshql, 2024-05-11:
> > 我这不用加也ok，ssh也连的上，重启了四五次也没问题
> 
> 我这边不加u-boot.emmc 无法开机 如果把extlinux.conf 删了可以开机，但是网卡不识别 看来我们的问题还不一样
> 
> U盘启动加了extlinux.conf 和 u-boot.ext 就一切正常 看来我们的问题还不一样 恭喜恭喜

你刷入emmc的时候有报错吗，我之前有一些报错，改了dst后就没报错了。
另外网卡好像和安卓底包有关，我用的是
https://github.com/ophub/kernel/releases/tag/tools 这里的底包
有些资料说是靠这个底包搞好的
[image]

安卓底包和改好的系统镜像我打包了一份，短接刷一次安卓，u盘烤录我这个镜像试试，分区都改好了直接写入emmc就行了https://www.123pan.com/s/0RVDVv-co6av.html提取码:ulRr

closed 2024-05-19 by ophub

mrsoshining, 2025-02-18:
b860AV2.1-A2试验中，已经不知道这是第几次尝试了，不同的版本、固件、内核、dtb

whmzsu, 2025-09-04:
> 老哥我这边成功了 [image] 不用替换内核啥的，就是要改一下频率，我改成1000....就ok了。刷入emmc，有线网卡也没问题（无线没试）。 步骤
> 
> 1. extlinux.conf.bak 改为extlinux.conf
> 2. u盘的boot分区下,复制u-boot-s905x-s912.bin这个文件，改成u-boot.ext
> 3. 替换自己改频率的dst
> 
> 这个是我修改后编译的dst： [meson-gxl-s905l2-x7-5g.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/15281730/meson-gxl-s905l2-x7-5g.zip)
> 
> 目前在Armbian_24.5.0_amlogic_s905l2_noble_6.1.87_server_2024.05.02.img.gz上测试没有问题了
> 
> github那个帖子都有提到，昨天没认真看，也要看看是不是emmc的芯片
> 
> 之前我一直不行估计是刷入emmc的时候频率太高导致断断续续的出现很多报错

怎么修改频率？