ophub/amlogic-s9xxx-openwrt #580: Which is correct release for NexBox A95X

dewgenenny, 2024-04-28:
Hey, apologies for potentially asking a basic question, however I'm struggling to figure out what is the correct image for flashing to a NexBox A95X device.

On the README page it lists "amlogic_s905x.img" as the image to use, however I can't find this image in the releases:

openwrt_official_amlogic_s905x-b860h_k5.10.215_2024.04.27.img.gz
openwrt_official_amlogic_s905x-b860h_k5.15.153_2024.04.08.img.gz

....
openwrt_official_amlogic_s905x-t95_k6.6.28_2024.04.21.img.gz
... 
etc

Any suggestions how to find the right one? Maybe @khancyr knows? 

Many thanks in advance for helping me 

[image]

ophub, 2024-04-28:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/make-openwrt/openwrt-files/common-files/etc/model_database.conf

211

dewgenenny, 2024-04-28:
> https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/make-openwrt/openwrt-files/common-files/etc/model_database.conf
> 
> 211

Awesome thanks.

I guess that means there's no pre-built binaries for this box?

closed 2024-05-19 by ophub