ophub/amlogic-s9xxx-openwrt #582: 能否编译一下小宝5.10.y内核的固件，6.1.y内核还是无法识别sata

phcl1998, 2024-05-09:
设备信息：小宝nas
 - SOC: [e.g. rk3399]

刚测试6.1.y内核无法识别sata，5.10.y可以

ophub, 2024-05-09:
你数的5.10是哪个系列的？stable还是rk35xx的呢？把你的终端界面那个地方截个图我看看内核的版本号是哪个。

phcl1998, 2024-05-09:
[image]
我之前使用的是stable里面5.10.212可以识别sata，更新6.1.87就没法识别sata。5.10的内核应该都可以识别sata

phcl1998, 2024-05-09:
我在线编译不出内核5.10.216的固件，能否编译一下5.10.216内核的固件我测试一下。

ophub, 2024-05-09:
你的这个是stable内核，https://github.com/ophub/kernel/releases/tag/kernel_stable

你在晶晨宝盒里可以更新内核，系统--晶晨宝盒--设置（内核选择5.10），在线下载更新--仅更新内核

phcl1998, 2024-05-09:
> 你的这个是stable内核，https://github.com/ophub/kernel/releases/tag/kernel_stable
> 
> 你在晶晨宝盒里可以更新内核，系统--晶晨宝盒--设置（内核选择5.10），在线下载更新--仅更新内核
仅更新内核之后无法启动

ophub, 2024-05-09:
不会吧，拔插电源试试
我上午把内核切换回5.10.y了，新编译的固件在releases里可以下载。

closed 2024-05-19 by ophub