ophub/amlogic-s9xxx-openwrt #585: 海信 hs530r

wlau29, 2024-05-14:
**Device Information | 设备信息**
- SOC: [rk3399]
- Model [海信 hs530r]
- 4g 运存
- 16Gemmc
- fn-link 6222d-ubb wifi
- rtl8211
- usb 2.0 *2
- usb 3.0 *2
- hdmi *1

**OpenWrt Version | 系统版本**
 - Kernel Version: 6.1.87
 - Release: Armbian_24.5.0_rockchip_hs530r_noble_6.1.87_server_2024.05.03.img.gz

**Describe the bug | 问题描述**
All issues will only remain open for one week to prioritize resolving them.
After that period, they will be closed but can still continue to be discussed in the thread.
所有的问题都只保留一周的开启状态供重点关注解决，超时将关闭，但仍然可以长期继续跟帖讨论。

Please provide a detailed description of the issue and, if necessary, attach a screenshot.
详细描述问题，并在必要时附上屏幕截图。

这块板的千兆网卡有问题。

```
perf3 -c 192.168.198.202   
Connecting to host 192.168.198.202, port 5201
[  5] local 192.168.198.103 port 44412 connected to 192.168.198.202 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   100 MBytes   841 Mbits/sec    0   3.01 MBytes       
[  5]   1.00-2.00   sec  96.2 MBytes   809 Mbits/sec    0   3.01 MBytes  
```
```
iperf3 -c 192.168.198.202 -R
Connecting to host 192.168.198.202, port 5201
Reverse mode, remote host 192.168.198.202 is sending
[  5] local 192.168.198.103 port 44408 connected to 192.168.198.202 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.01   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   1.01-2.01   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   2.01-3.01   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   3.01-4.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   4.00-5.00   sec  0.00 Bytes  0.00 bits/sec                  
[  5]   5.00-6.01   sec  0.00 Bytes  0.00 bits/sec  
```
用了 "ethtool -K eth0 tx off rx off" 没有帮助。

用了 "ethtool -s eth0 advertise 0xf" 设置成 100mbps, 就可以：

```
iperf3 -c 192.168.198.202 -R
Connecting to host 192.168.198.202, port 5201
Reverse mode, remote host 192.168.198.202 is sending
[  5] local 192.168.198.103 port 44444 connected to 192.168.198.202 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  11.2 MBytes  94.0 Mbits/sec                  
[  5]   1.00-2.00   sec  11.2 MBytes  94.1 Mbits/sec                  
[  5]   2.00-3.00   sec  11.2 MBytes  93.7 Mbits/sec   
```

ophub, 2024-05-14:
https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/armbian-files/common-files/etc/custom_service/start_service.sh

可以在自定义脚本里添加针对性的命令，开机自动执行。

这是通病还是个例？

wlau29, 2024-05-14:
> https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/armbian-files/common-files/etc/custom_service/start_service.sh
> 
> 可以在自定义脚本里添加针对性的命令，开机自动执行。
> 
> 这是通病还是个例？

offloading 是通病, 但不知道Linux 6 是否已经解决了。现在千兆用不了，还没找到方案
https://tadeubento.com/2019/nanopi-m4-rk3399-network-issues/
https://forum.armbian.com/topic/13544-nanopi-m4-and-other-rk3399-need-tcpudp-offloading-disabled/

wlau29, 2024-05-14:
我改了DTS 的TX delay, 从0x28 变成0x1f, 速度变正常了：

```
iperf3.exe -c 192.168.198.202
Connecting to host 192.168.198.202, port 5201
[  5] local 192.168.198.53 port 50348 connected to 192.168.198.202 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.01   sec  82.8 MBytes   689 Mbits/sec
[  5]   1.01-2.00   sec  80.2 MBytes   678 Mbits/sec
[  5]   2.00-3.01   sec  81.9 MBytes   678 Mbits/sec
[  5]   3.01-3.75   sec  59.8 MBytes   680 Mbits/sec

iperf3.exe -c 192.168.198.202 -R  -w 1m -M 1200
Connecting to host 192.168.198.202, port 5201
Reverse mode, remote host 192.168.198.202 is sending
[  5] local 192.168.198.53 port 50350 connected to 192.168.198.202 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   109 MBytes   908 Mbits/sec
[  5]   1.00-2.01   sec   110 MBytes   920 Mbits/sec
[  5]   2.01-3.01   sec   110 MBytes   921 Mbits/sec
[  5]   3.01-4.01   sec   109 MBytes   916 Mbits/sec

```
```
root@armbian:/boot/dtb/rockchip# diff -u rk3399-hs530r.dts.bak rk3399-hs530r.dts
--- rk3399-hs530r.dts.bak       2024-05-14 14:21:54.694976307 +0800
+++ rk3399-hs530r.dts   2024-05-14 14:05:23.624214517 +0800
@@ -291,7 +291,7 @@
                snps,reset-gpio = <0x1d 0x0f 0x01>;
                snps,reset-active-low;
                snps,reset-delays-us = <0x00 0x2710 0xc350>;
-               tx_delay = <0x28>;
+               tx_delay = <0x1f>;
                rx_delay = <0x11>;
                phandle = <0xb6>;
        };
```

ophub, 2024-05-14:
把你改好的这个打个压缩包上传上来，我更新到仓库里。

wlau29, 2024-05-14:
谢谢
[rk3399-hs530r.zip](https://github.com/ophub/amlogic-s9xxx-openwrt/files/15303968/rk3399-hs530r.zip)

closed 2024-05-14 by wlau29

ophub, 2024-05-16:
https://github.com/ophub/amlogic-s9xxx-armbian/commit/98640f1f19df5ca590a3644a881bf93b12f7d859

感谢分享，已经更新到仓库了。

496199544, 2024-09-01:
请问一下openwrt里面hs530r怎么系统概况没有温度显示