ophub/amlogic-s9xxx-openwrt #589: WEB可以打开，但是SSH连不上，总是报密码错误

goldsoft8888, 2024-05-19:
**Device Information | 设备信息**
 - SOC: [e.g. S095w]
 - Model [e.g. TX3 mini]

**OpenWrt Version | 系统版本**
 - Kernel Version: [e.g. 6.6.31]
 - Release: [e.g. official/lede/immortalwrt]

**Describe the bug | 问题描述**
All issues will only remain open for one week to prioritize resolving them.
After that period, they will be closed but can still continue to be discussed in the thread.
所有的问题都只保留一周的开启状态供重点关注解决，超时将关闭，但仍然可以长期继续跟帖讨论。

Please provide a detailed description of the issue and, if necessary, attach a screenshot.
详细描述问题，并在必要时附上屏幕截图。

openwrt_official_amlogic_s905w_k6.6.31_2024.05.18.img.gz
安装后，WEB可以打开，但是SSH连不上，总是报密码错误，更改为1234，或admin等都不行，

goldsoft8888, 2024-05-19:
Using username "root".
root@192.168.1.2's password:
Access denied
root@192.168.1.2's password:
Access denied
root@192.168.1.2's password:

ophub, 2024-05-19:
默认用户名：root
默认密码：password

goldsoft8888, 2024-05-19:
openwrt_official_amlogic_s905w_k5.15.156_2024.04.27  这个版本可以，我也是经常用openwrt了

goldsoft8888, 2024-05-19:
> 默认用户名：root 默认密码：password 默认SSH端口：22

是5月18号版本问题

closed 2024-05-19 by ophub