ophub/amlogic-s9xxx-openwrt #592: 关于内核不适配 问题

liaoke01, 2024-05-22:
**Device Information |  r2s
 - SOC: [rk3328]
 - Model [e.g. HK1]

**OpenWrt Version | 系统版本openwrt23.0.5
 - Kernel Version: [6.6.31]
 - Release: [e.g. official/lede/immortalwrt]

**Describe the bug | 问题描述**
在readme中明确表示rk3328设备支持所有内核，但是我在action build openwrt with image中。选择了最新的6.6内核进行构建，最后提示我无法使用该内核
[image]

ophub, 2024-05-22:
你可以`armbian-update -k 6.6`更新。
由于github空间不够用，我只保留了amlogic使用stable，可以生成多个内核的固件，rockchip和allwinner我都只生成了单个内核的固件。

liaoke01, 2024-05-22:
> 你可以`armbian-update -k 6.6`更新。 由于github空间不够用，我只保留了amlogic使用stable，可以生成多个内核的固件，rockchip和allwinner我都只生成了单个内核的固件。
> 
> 根据你的图片，我发现了个sha256sum的潜在错误，当没有img时会出现你上面的错误，我修复了。

我是在input中指定了单内核6.6.31版本，打包的时候提示不可用，我只能安装完6.1内核的版本之后在终端中在线更新吗？并且当我勾选6.6.x并且打开自动更新内核选项时，打包出来的镜像是6.1内核

ophub, 2024-05-23:
kernel_tags是stable的可以自由选择，其他的是指定了内核的不可自选，如果你想给这个盒子自选，fork仓库改model。目前我为了减少打包量，rockchip的都只打包了1个固定内核的。你只能安装后更新，不能打包时选择。

liaoke01, 2024-05-23:
> kernel_tags是stable的可以自由选择，其他的是指定了内核的不可自选，如果你想给这个盒子自选，fork仓库改model。目前我为了减少打包量，rockchip的都只打包了1个固定内核的。你只能安装后更新，不能打包时选择。

ok，请问可以添加对r2s的支持吗？

closed 2024-06-02 by ophub