ophub/amlogic-s9xxx-openwrt #598: 提点建议

phcl1998, 2024-05-27:
1.建议默认网络接口协议改为dhcp客户端，因为大部分盒子只有一个网口。以现在固件默认的静态ip还需要插盒子lan口连接电脑，进op管理界面（或者ssh）改网络模式为dhcp客户端才能插路由器上获取ip。
2.主题能否考虑增加Argon。
3.增加iStore商店。
[image]

[image]
[image]

basncy, 2024-05-31:
不建议上istore, 低性能设备带不动. (docker都不想要)

有线单网口改DHCP, 可以用以下内容替换 /etc/config/network

cat etc/config/network

`config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0'

config interface 'lan'
	option device 'br-lan'
	option proto 'dhcp'
	option peerdns '0'
	list dns '223.5.5.5'

config interface 'lan6'
	option proto 'dhcpv6'
	option reqprefix 'no'
	option sourcefilter '0'
	option delegate '0'
	option device 'br-lan'
	option reqaddress 'try'
	option peerdns '0'
	list dns '2400:3200::1'

`

phcl1998, 2024-05-31:
> 不建议上istore, 低性能设备带不动. (docker都不想要)
>
> 有线单网口改DHCP, 可以用以下内容替换 /etc/config/network
>
> cat etc/config/network
>
> `config interface 'loopback' option device 'lo' option proto 'static' option ipaddr '127.0.0.1' [...]

我意思在编译固件的时候修改网络模式，is只是个应用商店跟带不带的动没多大关系，你不要docker可以删除啊

closed 2024-06-01 by phcl1998