ophub/amlogic-s9xxx-openwrt #599: xiaobao op固件内核（6.1.87、6.6.32）还是无法识别sata

phcl1998, 2024-06-01:
**Device Information | 设备信息**
 - SOC: [e.g. rk3399]
 - Model [e.g. xiaobao]

**OpenWrt Version |    系统版本 ImmortalWrt 
 - Kernel Version: [e.g. 6.1.87]
 - Release: [e.g. official/lede/immortalwrt]

**Describe the bug | 问题描述6.1.87和6.6.32内核的openwrt固件还是无法识别sata
All issues will only remain open for one week to prioritize resolving them.
After that period, they will be closed but can still continue to be discussed in the thread.
所有的问题都只保留一周的开启状态供重点关注解决，超时将关闭，但仍然可以长期继续跟帖讨论。

Please provide a detailed description of the issue and, if necessary, attach a screenshot.
详细描述问题，并在必要时附上屏幕截图。

phcl1998, 2024-06-01:
本地打包xiaobao固件报错
[image]

phcl1998, 2024-06-01:
> 本地打包xiaobao固件报错 ![dabao](https://private-user-images.githubusercontent.com/125857645/335795630-9b2e19d5-c723-423b-bdb7-10ecb98d83f7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcyMzkzODgsIm5iZiI6MTcxNzIzOTA4OCwicGF0aCI6Ii8xMjU4NTc2NDUvMzM1Nzk1NjMwLTliMmUxOWQ1LWM3MjMtNDIzYi1iZGI3LTEwZWNiOThkODNmNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwMVQxMDUxMjhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iZGE2OTc0MWIzOWQ0MDVlOTRjNGJiZWU5YmJkYmM5MTI5ZTlmMmJhZThmYjI2ZDI4NzhlMzhkMzZiMWJjYzU0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Rhpn5wOArelscipcCsDul36JaaEWITO3neqFofwX2E8)

6.1.87和6.6.32内核无法识别sata，现在打包5.10内核也报错

ophub, 2024-06-01:
https://github.com/ophub/amlogic-s9xxx-armbian/issues/2335

你看看这个帖子，现在我设置成只打包6.6.y了。你可以改kernel_tags为stable就可以任意打包了。

phcl1998, 2024-06-01:
> [ophub/amlogic-s9xxx-armbian#2335](https://github.com/ophub/amlogic-s9xxx-armbian/issues/2335)
> 
> 你看看这个帖子，现在我设置成只打包6.6.y了。你可以改kernel_tags为stable就可以任意打包了。

感谢o大，更换上面链接里的dtb内核6.1.87已经成功识别sata

closed 2024-06-02 by ophub