ophub/amlogic-s9xxx-openwrt #600: 添加iStore提供的feed源就可以编译出istore首页的openwrt

phcl1998, 2024-06-07:
项目地址:https://github.com/linkease/nas-packages-luci
效果:
[image]
[image]
[image]
[image]
1.复刻ophub openwrt仓库，在diy-part1.sh脚本里添加
echo >> feeds.conf.default
echo 'src-git istore https://github.com/linkease/istore;main' >> feeds.conf.default
echo >> feeds.conf.default
echo 'src-git nas https://github.com/linkease/nas-packages.git;master' >> feeds.conf.default
echo 'src-git nas_luci https://github.com/linkease/nas-packages-luci.git;main' >> feeds.conf.default
2.在config文件里 Applications 下增加
# nas-packages-luci
CONFIG_PACKAGE_luci-app-ddnsto=y
CONFIG_PACKAGE_luci-app-diskman=y
CONFIG_PACKAGE_luci-app-istorex=y
CONFIG_PACKAGE_luci-app-linkease=y
CONFIG_PACKAGE_luci-app-quickstart=y
CONFIG_PACKAGE_luci-app-unishare=y
CONFIG_PACKAGE_luci-nginxer=y

closed 2024-07-13 by ophub

b051431, 2024-07-22:
编译刷机报软件源配置错误

b051431, 2024-07-22:
编译刷机报软件源配置错误

Vinke2, 2024-10-29:
用的是哪个源码官方 or lede or immortalWrt?