ophub/amlogic-s9xxx-openwrt #601: dg3399如何调整固件分区大小,默认只使用了1G

mazige1997, 2024-06-08:
**Device Information | 设备信息**
 - SOC: rk3399
 - Model dg3399

**OpenWrt Version | 系统版本**
 - Kernel Version: 6.1.92
 - Release: [e.g. immortalwrt

**Describe the bug | 问题描述**
发现在项目目录下改动root_mb，还是只用了1G，以dg3399 2+8G为例，应该如何修改呢，使刷入固件能匹配rom空间

mazige1997, 2024-06-08:
[image]

ophub, 2024-06-09:
看仓库首页文档，可以使用-s自定义固件大小，但是太小了放不下，1G以内应该是不够的。

mazige1997, 2024-06-09:
> 看仓库首页文档，可以使用-s自定义固件大小，但是太小了放不下，1G以内应该是不够的。

在这个仓库改哪里才能实现自定义固件大小呢

mazige1997, 2024-06-09:
> 看仓库首页文档，可以使用-s自定义固件大小，但是太小了放不下，1G以内应该是不够的。
是改这个项目库下的make文件里面的root_mb值嘛？？？

ophub, 2024-06-09:
可以不用改文件，编译的使用使用`-m boot_mb/root_mb`

mazige1997, 2024-06-09:
> 可以不用改文件，编译的使用自定义大小的参数：`-s boot_mb/root_mb`

利用这个项目库github action编译固件这个自定义参数在哪添加还是说还是要改动make文件里的boot_mb、root_mb？另外，我想问一下，比如说dg3399是有8G的rom，boot_mb+root_mb=8G嘛？

ophub, 2024-06-09:
不是boot_mb+root_mb=8G，这2个参数设置的大小，只要能把文件放进去即可，和实际的盒子无关，一个固件往往可以在很多设备上通用。

你如果对改rebuild比较好理解，就改这个文件吧。
这个文件里的默认设置是通用的，可以自定义的参数都在首页的文档里有详细的说明了，本地编译和Actions编译类似，你可以在Actions里添加armbian_size，例如：

```
- name: Rebuild Armbian
  uses: ophub/amlogic-s9xxx-armbian@main
  with:
    build_target: armbian
    armbian_path: build/output/images/*.img
    armbian_board: dg3399
    armbian_kernel: 6.1.y
    armbian_size: 512/2560
```

mazige1997, 2024-06-09:
好的，谢谢指导

closed 2024-06-09 by mazige1997