ophub/amlogic-s9xxx-openwrt #602: reduced RAM in X96 Max+ A100 -- dtb mistake

taras-filatov, 2024-06-09:
**Device Information | 设备信息**
 - SOC: S095x3
 - Model: X96 Max+ A100 (4G RAM, 64G flash, 100Mb ethernet)

Issue about corresponding dtb.
From model_database.conf:
my model: 505     :X96-Max+_A100
my SoC: s905x3
my fdt file: meson-sm1-sei610.dtb

I noticed, that in openwrt/armbian with that dtb (meson-sm1-sei610.dtb) all running good, but available ram less than 1GB.
Hardware -- 4GB

in meson-sm1-sei610.dtb decompiled memory description:
```
        #address-cells = <0x02>;
        #size-cells = <0x02>;
        memory@0 {
                device_type = "memory";
                reg = <0x00 0x00 0x00 0x40000000>;
        };

```
thus limiting ram to 1073741824 bytes, 1G.

I extract and decompile dtb from firmware. There are 3 dtb's in one, Amlogic multi dtb. Bootloader make desision while boot and select variant 2 (from 0-1-2):
```
      Amlogic multi-dtb tool
      GZIP format, decompress...
      Multi dtb detected
      Multi dtb tool version: v2 .
      Support 3 dtbs.
        aml_dt soc: sm1 platform: ac213 variant: 4g
        dtb 0 soc: g12a   plat: u212   vari: 4g
        dtb 1 soc: sm1   plat: ac213   vari: 2g
        dtb 2 soc: sm1   plat: ac213   vari: 4g
      Find match dtb: 2

```

In that variant:
```
        model = "Amlogic";
        amlogic-dt-id = "sm1_ac213_4g";
        compatible = "amlogic, g12a";
        interrupt-parent = <0x01>;
        #address-cells = <0x01>;
        #size-cells = <0x01>;
...
        memory@00000000 {
                device_type = "memory";
                linux,usable-memory = <0x100000 0xf0800000>;
        };
...
```

Same way of memory description is in other dtb's in this bundle, and in dtb's from official armbian, LibreELEC, CoreELEC: defined only linux,usable_memory (memory reservation for kernel recovery purposes, afaik)

And only in device tree from running Android system (not extracted from blob, but from runnning /proc filesystem)
 we see info about ram size:
```
        memory@00000000 {
                device_type = "memory";
                linux,usable-memory = <0x100000 0xf0800000>;
                reg = <0x00 0xd8000000>;
        };

```
It looks lite there is no need to hardcode RAM size in dtb -- system add it for application usage. All difference withing dtb's now is linux,usable-memory. in meson-sm1-sei610.dtb it equals to my orig dtb. But actually it can be ommited as I see no realisation of kernel recovery....

I propose corrected dtb -- without limit of RAM (reg record), hope you add it to build. 

[meson-sm1-sei610-fullram.dtb.gz](https://github.com/user-attachments/files/15751218/meson-sm1-sei610-fullram.dtb.gz)
[meson-sm1-sei610-fullram.dts.gz](https://github.com/user-attachments/files/15751219/meson-sm1-sei610-fullram.dts.gz)

ophub, 2024-06-09:
https://github.com/unifreq/linux-6.1.y/blob/main/arch/arm64/boot/dts/amlogic/meson-sm1-sei610.dts

https://github.com/unifreq/linux-6.1.y/blob/main/arch/arm64/boot/dts/amlogic/meson-sm1-x96-max-plus-q2.dts

The `meson-sm1-sei610.dts` is a generic file provided by the kernel source, and it is used by many DTS applications. It is not recommended to modify it directly. Instead, you can generate a new DTS file based on this one. For example, as shown in `meson-sm1-x96-max-plus-q2.dts`, you can include an existing file and rewrite the sections that differ to create a new DTS file with a different name. It is recommended to create a new file in this manner, perhaps naming it `meson-sm1-x96-max-a100.dts`

ophub, 2024-06-09:
[meson-sm1-x96-max-plus-a100.zip](https://github.com/user-attachments/files/15751486/meson-sm1-x96-max-plus-a100.zip)

[image]

I tested the compilation, and it works fine. You can extract the files and upload the DTB file to the /boot/dtb/amlogic/ directory of the Armbian system. Then, modify the DTB name in `/boot/uEnv.txt` to `meson-sm1-x96-max-plus-a100.dtb`, and try rebooting.

taras-filatov, 2024-06-09:
yep... "#include" is good idea, when working in large source tree, like openwrt. I do so there while building for modded tplinks.
But vanilla dtc (from my debian, or maked by me now from git.kernel.org source) cannot cope with #include directive in any way...

Oh, I find it, dts with #include should be preprocessed first with cpp (hope, you know it, I had not)
[](https://stackoverflow.com/questions/50658326/device-tree-compiler-not-recognizes-c-syntax-for-include-files)

```
cpp -nostdinc -I include -I arch  -undef -x assembler-with-cpp meson-sm1-x96-max-plus-a100.dts meson-sm1-x96-max-plus-a100.dts.preprocessed
dtc -I dts -O dtb -o meson-sm1-x96-max-plus-a100.dtb meson-sm1-x96-max-plus-a100.dts.preprocessed

```
And in any case file you send (dtb) still has limit to 1G.
And it's because that in orde to vanish reg parameter from memory@0 clause original clause (memory@0) shold be deleted first with /delete-node/. So we have:
```
// SPDX-License-Identifier: (GPL-2.0+ OR MIT)
/*
 * Copyright (c) 2024 taras-filatov, unifreq.
 */

/dts-v1/;

#include "meson-sm1-sei610.dts"

/ {
        compatible = "x96-max-a100", "amlogic,sm1";
        model = "X96 MAX+ A100";

/delete-node/ memory@0;

        memory@0 {
                device_type = "memory";
                // reg = <0x0 0x0 0x0 0x40000000>;
                linux,usable-memory = <0x0 0x100000 0x0 0xf0800000>;
        };
};
```
[meson-sm1-x96-max-plus-a100-corrected.zip](https://github.com/user-attachments/files/15754011/meson-sm1-x96-max-plus-a100-corrected.zip)

ophub, 2024-06-10:
[a100.zip](https://github.com/user-attachments/files/15756314/a100.zip)

try again.

taras-filatov, 2024-06-10:
all fine now. Thank you!

ophub, 2024-06-11:
Take a screenshot of the terminal information page where the correctly recognized memory size can be seen.

Take photos of the box, both front and back, and share them.

Does your A100 have an Android system? If so, please provide a download link, and I will save a copy.

ophub, 2024-06-11:
Thank you for sharing, the file has been added to the upstream kernel repository.

https://github.com/unifreq

taras-filatov, 2024-06-11:
Photos... Actually my model is "X96Max+ A100", and really corrrespond to described in mode_database.inf, and looks like on photos here [](https://github.com/ophub/amlogic-s9xxx-armbian/issues/779)
but ok...
[image]
[image]
[image]
[image]

taras-filatov, 2024-06-11:
Screenshots -- harder, my hdmi is burned out, and I still do not know, how to fire up cvbs on armbian/openwrt. I hope you're really do not want them, but info from device.  So I attach logs with android boot and with openwrt boot.
[android.log](https://github.com/user-attachments/files/15789809/android.log)
[openwrt.log](https://github.com/user-attachments/files/15789810/openwrt.log)

taras-filatov, 2024-06-11:
don't close thread, I'll check one issue...

taras-filatov, 2024-06-11:
Oh! I see, source dtb (meson-sm1-sei610.dtb) is changed. In compare with mine, from your pre-latest release. The main issue is ok -- memory limit removed.
But now besides other things, changed paths to wifi devices.... But warnings of dts while decompilng is reduced too.

taras-filatov, 2024-06-12:
So that's all, problem solved. Thank you!

closed 2024-06-12 by taras-filatov