ophub/amlogic-s9xxx-openwrt #603: s905lb r3300l-k5.10.218  报Failed to connect to ubus

x2358473, 2024-06-11:
**Device Information | 设备信息**
 - SOC: [e.g. S095lb]
 - Model [e.g. b860av1.1-t]

**OpenWrt Version | 系统版本**
 - Kernel Version: [e.g. 5.10.218]
 - Release: [e.g.lede]

**Describe the bug | 问题描述**
使用6.1内核的版本只有788M内存。降级到5.10.218后开机会报Failed to connect to ubus
使用5.10.y内核编译lede固件也是报Failed to connect to ubus
目前除了5.10内核可以929M内存。其它都只有700多M.
在6.1内核中替换u-boot也只有700多M

ophub, 2024-06-11:
这个设备目前不在已知列表里，你是第一个人，需要反复修改配置信息，找到合适的组合。

调整dtb的内存识别方法：
https://github.com/ophub/amlogic-s9xxx-openwrt/issues/602

u-boot添加方法看12.12:
https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/documents/README.cn.md

参考上面的方法，添加u-boot或者调整dtb

x2358473, 2024-06-11:
> 这个设备目前不在已知列表里，你是第一个人，需要反复修改配置信息，找到合适的组合。
> 
> 调整dtb的内存识别方法： #602
> 
> u-boot添加方法看12.12: https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/documents/README.cn.md
> 
> 参考上面的方法，添加u-boot或者调整dtb

感谢，剩下的我慢慢去调整。

closed 2024-06-11 by x2358473