ophub/amlogic-s9xxx-openwrt #604: seewo SV21 新提取的dts和dtb文件，能帮忙辛苦添加试试么

itoywh, 2024-06-14:
您好，之前有反馈过sv21现在的dts有问题，cpu只能跑在最低频率上（所以跑分只有1w2左右，实际RK3568是能跑到2w7左右）

之前的反馈贴：https://github.com/ophub/amlogic-s9xxx-openwrt/issues/586

这次特意从该机器的原始安卓固件中提取了dts和dtb文件，能帮忙辛苦添加重新编译试试么

[1.zip](https://github.com/user-attachments/files/15832275/1.zip)

ophub, 2024-06-14:
根据你提供的安卓dts，他的工作频率是固定1992000000赫兹（即1.992 GHz），工作电压为1150000微伏（即1.15伏）
```
	cpu0-opp-table {
		compatible = "operating-points-v2";
		opp-shared;
		rockchip,temp-hysteresis = <0x1388>;
		rockchip,low-temp = <0x0>;
		rockchip,low-temp-adjust-volt = <0x450 0x648 0x124f8>;
		nvmem-cells = <0x6>;
		nvmem-cell-names = "leakage";
		phandle = <0x3>;

		opp-1992000000 {
			opp-hz = <0x0 0x76bb8200>; //1992000000
			opp-microvolt = <0x118c30 0x118c30 0x118c30>; //1150000
			clock-latency-ns = <0x9c40>;
		};
	};
```

我把主线内核里seewo sv21也参照他的固定了，你下载，解压，覆盖armbian的/boot/dtb/rockchip/目录的同名文件，重启试试。

itoywh, 2024-06-14:
> 根据你提供的安卓dts，他的工作频率是固定1992000000赫兹（即1.992 GHz），工作电压为1150000微伏（即1.15伏）
> 
> ```
> 	cpu0-opp-table {
> 		compatible = "operating-points-v2";
> 		opp-shared;
> 		rockchip,temp-hysteresis = <0x1388>;
> 		rockchip,low-temp = <0x0>;
> 		rockchip,low-temp-adjust-volt = <0x450 0x648 0x124f8>;
> 		nvmem-cells = <0x6>;
> 		nvmem-cell-names = "leakage";
> 		phandle = <0x3>;
> 
> 		opp-1992000000 {
> 			opp-hz = <0x0 0x76bb8200>; //1992000000
> 			opp-microvolt = <0x118c30 0x118c30 0x118c30>; //1150000
> 			clock-latency-ns = <0x9c40>;
> 		};
> 	};
> ```
> 
> 现在 [rk3568-seewo-sv21.dts](https://github.com/unifreq/linux-6.1.y/blob/be15d5aa1f210da0716c1d691b7b300b8e288c3c/arch/arm64/boot/dts/rockchip/rk3568-seewo-sv21.dts#L8) 使用的主线内核的频率和电压设置 [rk3568.dtsi](https://github.com/unifreq/linux-6.1.y/blob/be15d5aa1f210da0716c1d691b7b300b8e288c3c/arch/arm64/boot/dts/rockchip/rk3568.dtsi#L249-L254)：
> 
> ```
> &cpu0_opp_table {
> 	opp-1992000000 {
> 		opp-hz = /bits/ 64 <1992000000>;
> 		opp-microvolt = <1150000 1150000 1150000>;
> 	};
> };
> ```
> 
> 两个的CPU设置频率和电压完全相同，都是固定的1.992 GHz和1.15伏电压。CPU跑分和这个设置有关。 你看看`lscpu`现在的频率。如果频率不对，在armbian里的 `/etc/default/cpufrequtils` 默认设置最大频率，最新频率你可以手动固话了试试。

您看下这个反馈帖子：https://github.com/ophub/amlogic-s9xxx-openwrt/issues/586
用lscpu后，是根本就显示不了运行频率。正常的情况下应该有个最大频率、最小频率的显示

ophub, 2024-06-14:
[image]

正常情况下应该能看到，可能dts有什么地方写的不对？

closed 2024-07-13 by ophub