ophub/amlogic-s9xxx-openwrt #612: Chainedbox-L1-Pro 刷入无法连接网络

0xffffharry, 2024-06-30:
**Device Information | 设备信息**
 - SOC: rk3328
 - Model: Chainedbox-L1-Pro

**OpenWrt Version | 系统版本**
 - Kernel Version: 5.10.220
 - Release: official 23.05.3 | ImmortalWrt 23.05.2

**Describe the bug | 问题描述**
All issues will only remain open for one week to prioritize resolving them.
After that period, they will be closed but can still continue to be discussed in the thread.
所有的问题都只保留一周的开启状态供重点关注解决，超时将关闭，但仍然可以长期继续跟帖讨论。

Please provide a detailed description of the issue and, if necessary, attach a screenshot.
详细描述问题，并在必要时附上屏幕截图。

刷入启动后，无法获取IP地址，网口灯不亮，Windows下卡在正在识别...

0xffffharry, 2024-06-30:
OpenWrt 通过 downloads.openwrt.org 直接下载 rootfs.tar.gz 打包，或者使用 ImageBuilder 打包获取 rootfs.tar.gz

ophub, 2024-07-01:
https://github.com/ophub/amlogic-s9xxx-openwrt/releases

试试这里下载的可以不可以

0xffffharry, 2024-07-01:
> https://github.com/ophub/amlogic-s9xxx-openwrt/releases
> 
> 试试这里下载的可以不可以

试过，不行

closed 2024-07-19 by 0xffffharry

bowen-shiyong752, 2025-04-10:
> 已解决，新版本OpenWrt固件将网口`eth0`识别为`end0`，需要手动修改，或者在构建固件时手动更正

老师，请帮忙，同样问题。出个教程。十分感谢！

0xffffharry, 2025-04-11:
imagebuilder自己打包，打包前更改/etc/config/network，把网口更改过来

Javebean, 2025-04-12:
@0xffffharry @bowen-shiyong752 老师 我也是同样的问题 但是没有找到end0 这个错误配置
我用的是action打包的 您用的是本地打包吗？

Javebean, 2025-04-12:
@ophub @hex-ci @0xffffharry @bowen-shiyong752 请教大家