ophub/amlogic-s9xxx-openwrt #616: 使用晶晨宝盒安装openwrt至emmc报错

Kryo123456, 2024-07-08:
**Device Information | 设备信息**
 - SOC: S095d
 - Model: Phicomn N1

**OpenWrt Version | 系统版本**
 - Kernel Version: 6.6.37
 - Release: immortalwrt

**Describe the bug | 问题描述**
使用u盘能正常启动，但是使用晶晨宝盒将固件安装到emmc时提示“/boot/dtb/amlogic does not exist”,此外似乎防火墙里面没有开启fullcone nat选项，这个问题看到issue有人提了。

413137366, 2024-07-16:
Device Information | 设备信息

SOC: S905l3a
Model: M401A
OpenWrt Version | 系统版本

Kernel Version: 6.6.36、6.6.39
Release: official
Describe the bug | 问题描述
同上！

413137366, 2024-07-16:
更换lede 6.6.39版本后，可以安装到emmc，但开机卡在logo屏。

413137366, 2024-07-16:
找到之前存档的  6.6.31，使用正常

zhongyingbiao, 2024-07-17:
一样的问题 解决了吗

bzzjh, 2024-07-17:
openwrt_immortalwrt_amlogic_s905d_k6.1.98_2024.07.15.img

bzzjh, 2024-07-17:
> 找到之前存档的 6.6.31，使用正常

有这个版本吗，我似乎没找到，可以给个链接吗，多谢！

413137366, 2024-07-18:
> > 找到之前存档的 6.6.31，使用正常
> 
> 有这个版本吗，我似乎没找到，可以给个链接吗，多谢！

6.6.31版本是我几个月之前下载存在电脑里的。
我在releases下载了多个版本进行测试，2024年6月初编译6.1版本内核的固件是正常的。大家可以下载试试。

ophub, 2024-07-18:
系统  >  晶晨宝盒  >  在线下载更新  >  仅更新晶晨宝盒
把插件升级到最新版，这个安装错误是因为固件里没有perl导致的，我已经把perl命令换成shell语法了。

bzzjh, 2024-07-18:
> 系统 > 晶晨宝盒 > 在线下载更新 > 仅更新晶晨宝盒 把晶晨插件升级到最新版，这个安装错误是因为固件里perl无法正常执行导致的，我已经把perl命令换成shell语法了。

在192.168.1.1 U盘启动里配置好网络，点在线更新，系统直接挂了，无法访问WEB页面，重启也无法进入U盘里的openwrt了

ophub, 2024-07-18:
[image]
[image]
[image]

bzzjh, 2024-07-18:
多谢！重新下载了 openwrt_official_amlogic_s905d_k6.6.39_2024.07.15.img，然后再重复操作就可以了，感谢！

> ![Snip20240718_1](https://private-user-images.githubusercontent.com/68696949/349776171-5a8b8bbe-b959-4c34-ae26-ba067325967f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEyNzU2NTYsIm5iZiI6MTcyMTI3NTM1NiwicGF0aCI6Ii82ODY5Njk0OS8zNDk3NzYxNzEtNWE4YjhiYmUtYjk1OS00YzM0LWFlMjYtYmEwNjczMjU5NjdmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzE4VDA0MDIzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMyNDdiYjExYWM2NTRmZGQ1OTVkODEyYzhiMTIxY2VlMWVkMzhkOTg5YTU1OGQ4MDQ4NjAxMGU4Mzc4YWZmZGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.YxLT6xFwwBSDzFu-DQiENcPDcavj76o5N4260q6LFBk) ![Snip20240718_2](https://private-user-images.githubusercontent.com/68696949/349776182-fad8d839-00e5-46f6-8201-05b952a3c831.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEyNzU2NTYsIm5iZiI6MTcyMTI3NTM1NiwicGF0aCI6Ii82ODY5Njk0OS8zNDk3NzYxODItZmFkOGQ4MzktMDBlNS00NmY2LTgyMDEtMDViOTUyYTNjODMxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzE4VDA0MDIzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ3NTNmY2JjNzdlZTE2M2ZjYjYwZWRlNGE2M2RhMjYzYjZkOGQyN2Q0NmI2NGZlNDk2Y2MwNTYwNjA3MmI4MTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.CrC4I-RXYQ07GcuRDSY5BBOdvAqFIG1fpX1v2LbWVeQ) ![Snip20240718_3](https://private-user-images.githubusercontent.com/68696949/349776187-140333b7-b85b-4b1f-9b2c-1c9bad2434cd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjEyNzU2NTYsIm5iZiI6MTcyMTI3NTM1NiwicGF0aCI6Ii82ODY5Njk0OS8zNDk3NzYxODctMTQwMzMzYjctYjg1Yi00YjFmLTliMmMtMWM5YmFkMjQzNGNkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzE4VDA0MDIzNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk1MmQ5NTZmZTUzZjZlMTg2MjkzYWJjMWRlZjdmNDQ4MmZiOTI5ZmU0YWU1NWRkNzU2Yjk1NTZmMmQwNjI5ZjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RrXi-AmQ-ro448zhzBgOEY-yitfrB5fWWtvu4Q-AnSM)

bzzjh, 2024-07-18:
> ![Snip20240718_1](https://private-user-images.githubusercontent.com/68696949/349776171-5a8b8bbe-b959-4c34-ae26-ba067325967f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXV [...]

感谢，已经搞定了！！

bzzjh, 2024-07-18:
@ophub 
新的问题出现了。我安装Openclash，出现以下错误，似乎是内核版本不匹配，但是openclash好像不是基于稳定版本的内核，那个哈希值的内核版本哪里去找呢？

`Installing luci-app-openclash (0.46.014-rbeta) to root...
Downloading https://mirrors.vsean.net/openwrt/snapshots/packages/aarch64_generic/luci/luci-app-openclash_0.46.014-rbeta_all.ipk`

`Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-nfnetlink
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-nf-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-nf-conntrack-netlink
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-nf-reject6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-nf-conntrack6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-nf-nat
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-nf-log
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-nf-log6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-crypto-hash
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-crypto-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-lib-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-nft-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-inet-diag
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-nf-tproxy
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-nft-tproxy
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-tproxy found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1) for kmod-tun
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-openclash:
 * 	kernel (= 6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1)
 * opkg_install_cmd: Cannot install package luci-app-openclash.`

bzzjh, 2024-07-18:
我手动升级到稳定版本内核 6.6.39-ophub 也不行，貌似得找到这个版本的内核：6.6.39~4ad8919e6cb358ce9faee3c576171c3d-r1

Kryo123456, 2024-07-18:
> @ophub 新的问题出现了。我安装Openclash，出现以下错误，似乎是内核版本不匹配，但是openclash好像不是基于稳定版本的内核，那个哈希值的内核版本哪里去找呢？
>
> `Installing luci-app-openclash (0.46.014-rbeta) to root... Downloading [...]

我也遇到了这个问题。

bzzjh, 2024-07-18:
> 我也遇到了这个问题。

我查了，也就是必须自己编译，release里的版本就算安装了也安装不了任何软件！所有的软件都是依赖错误

ophub, 2024-07-18:
自己编译最新固件，然后完整更新固件

closed 2024-07-18 by ophub

413137366, 2024-07-18:
我昨天要安装无线网卡驱动，刚遇到类似情况，我的做法是先下载6.6.40内核包升级，然后在openwrt镜像库下载kernel6.6.40手动上传安装即可

不懂原理，可以一试