ophub/amlogic-s9xxx-openwrt #620: tvbox设备 zte zxv10 b860AV1.1-T (nand闪存) 应该选择什么版本？

UiLgNoD-lIaMtOh, 2024-07-29:
# 问题
    想刷 openwrt ，找了很多教程，在 github 只有这里可以支持 s905 系列的芯片，应该选择什么版本的 openwrt 固件且支持 rtl8189ftv 无线？
    我知道这很难，也弄了很久，我不是个愿意麻烦人的人，但是我没办法了，只能来求了，如果需要我自己编译，能否告诉我，我应该怎么做，在硬件嵌入式领域，完全不知所措，就像不知道怎么走路的人，我缺乏太多的知识，希望能帮助我，会花费你们的时间，我很抱歉，我把能提供的所有信息，全部编辑出来在下面，来节省你们解决问题的时间，再次感谢

# tvbox设备
    型号 zte zxv10 b860AV1.1-T 
    无线 rtl8189ftv
    CPU 可能是 CPU为 S905m2-b

## CPU 参考依据: 
    https://www.znds.com/tv-1246490-1-1.html

# 刷机引导其他系统准备过程: 
    0. 解压 `111-电信版本中兴B860AV1.1T-S905M-B-NAND闪存-当贝桌面线刷固件包（亲测）.zip` 得到 `刷机包.img`
    1. 安装启动工具 USB_Burning_Tool v2.05.15 加载 `刷机包.img` 固件
    2. 拆机短接 nand 左侧第6和第7个引脚，看到 USB_Burning_Tool 有连接反应 1% 时断开短接等待完成
    3. 解压 `2.20191218-Q7-4.4.2-root-twrp-Milton.rar` 得到 `20191218-Q7-4.4.2-root-twrp-Milton.img`
    4. 使用 USB_Burning_Tool v2.05.15 加载 `20191218-Q7-4.4.2-root-twrp-Milton.img` 固件
    5. 拆机短接 nand 左侧第6和第7个引脚，看到 USB_Burning_Tool 有连接反应 1% 时断开短接等待完成
    6. 开机获得 root 权限，安装 `Reboot to LibreELEC_1.1_Apkpure.apk` 获得引导启动其它系统的能力

## 固件来源
0. `111-电信版本中兴B860AV1.1T-S905M-B-NAND闪存-当贝桌面线刷固件包（亲测）.zip`及线刷工具来源：
    链接: https://pan.baidu.com/s/1u3S2ukNMIigbuPyLMNyd8w
    提取码: a2sx
    刷机后-盒子设置密码：6321
1. `20191218-Q7-4.4.2-root-twrp-Milton.img` 和 `Reboot to LibreELEC_1.1_Apkpure.apk` 来源: 
    https://drive.google.com/drive/u/1/folders/11GwfUoTW-Rz8qu78GIKJILhCIA7kmNWX

# 以下是本设备的完整外观图和主板图，以及 无线网卡 和 nand 闪存局部图

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

[image]

再次感谢

UiLgNoD-lIaMtOh, 2024-07-30:
经过这一天的研究，会使用 actions 了虽然只有一部分，我发现
内核为 5.15.y 
board device 为 s905x-b860h
编译出的固件只是能识别 wifi 设备，不能开热点也不能连wifi
我再试试把内核降低看看能不能用
[image]

ophub, 2024-07-30:
先测试有线的网络可能正常使用。选择对dtb就可以

UiLgNoD-lIaMtOh, 2024-07-30:
> 先测试有线的网络可能正常使用。选择对dtb就可以

有线网络确实正常，但 dtb 都试过了，没有一个可以让 wifi 功能正常，我再继续降级内核尝试 s905mb 吧，虽然我不知道 s905mb 是不是就是 s905m-b

ophub, 2024-07-30:
https://github.com/unifreq/linux-6.6.y/tree/main/drivers/net/wireless/realtek/rtl818x

无线没驱动，不用测。用有线吧。

UiLgNoD-lIaMtOh, 2024-07-30:
> https://github.com/unifreq/linux-6.6.y/tree/main/drivers/net/wireless/realtek/rtl818x
> 
> 无线没驱动，不用测。 用有线吧。 s905m-b就是s905mb

好的，谢谢你，那我关闭问题