ophub/amlogic-s9xxx-openwrt #621: github actions在线编译运行时选择内核5.15.y，在最后打包时下载使用依然是最新的6.6.y内核

LemonCrab666, 2024-08-11:
**Device Information | 设备信息**
 - SOC: RK3399
 - Model dg3399

**OpenWrt Version | 系统版本**
 - Kernel Version: 5.15.162
 - Release: immortalwrt

**Describe the bug | 问题描述**
github actions在线编译运行时选择内核5.15.y，在最后打包时下载使用依然是最新的6.6.y内核

[image]

运行任务时的选项如下
[image]

ophub, 2024-08-11:
model固定了，rk3399不支持5.15.y

LemonCrab666, 2024-08-11:
> model固定了，rk3399不支持5.15.y

手动换5.15也是能用的……

ophub, 2024-08-12:
rockchip系列从开始使用的时候就是使用的6.1.y，今年加了6.6.y，有一些使用的是专用的linux-rk35xx/5.10.y内核，但是从来没在5.15.y下使用过。

这个5.15.y的内核源码里缺少很多dts和针对性优化，不推荐使用，如果你觉得可以使用就改下model自己添加其他内核吧。for仓库并自己修改model，设置成你喜欢的内核。设置方法见此文件头部的说明。

https://github.com/ophub/amlogic-s9xxx-armbian/blob/main/build-armbian/armbian-files/common-files/etc/model_database.conf

LemonCrab666, 2024-08-12:
> rockchip系列从开始使用的时候就是使用的6.1.y，今年加了6.6.y，有一些使用的是专用的linux-rk35xx/5.10.y内核，但是从来没在5.15.y下使用过。
>
> 这个5.15.y的内核源码里缺少很多dts和针对性优化，不推荐使用，如果你觉得可以使用就改下model自己添加其他内核吧。for仓库并自己修改model，设置成你喜欢的内核。设置方法见此文件头部的说明。 [...]

了解了，多谢🙏

aguoxing, 2024-08-21:
> rockchip系列从开始使用的时候就是使用的6.1.y，今年加了6.6.y，有一些使用的是专用的linux-rk35xx/5.10.y内核，但是从来没在5.15.y下使用过。
>
> 这个5.15.y的内核源码里缺少很多dts和针对性优化，不推荐使用，如果你觉得可以使用就改下model自己添加其他内核吧。for仓库并自己修改model，设置成你喜欢的内核。设置方法见此文件头部的说明。 [...]

选6.1.y也不行吗
`Run ophub/amlogic-s9xxx-openwrt@main
  with:
    openwrt_path: openwrt/output/*rootfs.tar.gz
    openwrt_board: emb3531
    openwrt_kernel: 6.1.y
    auto_kernel: false
    kernel_repo: ophub/kernel
    kernel_usage: stable
    builder_name: yourname
    mode: ophub
  env:
    FEEDS_CONF: config/lede-master/feeds.conf.default
    CONFIG_FILE: config/lede-master/config
    DIY_P1_SH: config/lede-master/diy-part1.sh
    DIY_P[2](https://github.com/aguoxing/amlogic-s9xxx-openwrt/actions/runs/10483811454/job/29037134348#step:14:2)_SH: config/lede-master/diy-part2.sh
    TZ: America/New_York
Run cd ${GITHUB_ACTION_PATH}
ophub package actions path: [ /home/runner/work/_actions/ophub/amlogic-s9xxx-openwrt/main ]
Get openwrt file input parameters: [ openwrt/output/*rootfs.tar.gz ]
Copy OpenWrt rootfs file: [ openwrt/output/*rootfs.tar.gz ]
'/home/runner/work/amlogic-s9xxx-openwrt/amlogic-s9xxx-openwrt/openwrt/output/openwrt-armvirt-64-generic-rootfs.tar.gz' -> 'openwrt-armvirt/openwrt-armvirt-64-generic-rootfs.tar.gz'
About the [ openwrt-armvirt ] directory: 
total 178592
-rw-r--r-- 1 runner docker 182870[3](https://github.com/aguoxing/amlogic-s9xxx-openwrt/actions/runs/10483811454/job/29037134348#step:14:3)03 Aug 21 02:56 openwrt-armvirt-64-generic-rootfs.tar.gz
Start to make openwrt...
[ STEPS ] Welcome to make OpenWrt!
[ INFO ] Server running on Ubuntu: [ Release: jammy / Host: x86_6[4](https://github.com/aguoxing/amlogic-s9xxx-openwrt/actions/runs/10483811454/job/29037134348#step:14:4) ] 

[ STEPS ] Start Initializing Variables...
[ INFO ] The kernel tags list: [ stable ]
[ STEPS ] Start searching for OpenWrt file...
[ INFO ] OpenWrt file: [ openwrt-armvirt-[6](https://github.com/aguoxing/amlogic-s9xxx-openwrt/actions/runs/10483811454/job/29037134348#step:14:6)4-generic-rootfs.tar.gz ]
[ INFO ] The source_codename: [ _lede ], OpenWrt source code repository: [ github.com/coolsnowwolf/lede ]
[ STEPS ] Start downloading dependency files...
[ INFO ] firmware download completed.
[ INFO ] different-files download completed.
[ INFO ] balethirq download completed.
[ INFO ] app/sbin download completed.
[ INFO ] app/share download completed.
[ STEPS ] Start downloading the kernel files...
[ INFO ] (1.1) [ stable - 6.6.y ] Kernel download from [ https://github.com/ophub/kernel/releases/download/kernel_stable/6.6.y.tar.gz ]
curl: (22) The requested URL returned error: 404
curl: (22) The requested URL returned error: 404
curl: (22) The requested URL returned error: 404
curl: (22) The requested URL returned error: 404
curl: (22) The requested URL returned error: 404
curl: (22) The requested URL returned error: 404
curl: (22) The requested URL returned error: 404
curl: (22) The requested URL returned error: 404
curl: (22) The requested URL returned error: 404
curl: (22) The requested URL returned error: 404
tar (child): /home/runner/work/_actions/ophub/amlogic-s[9](https://github.com/aguoxing/amlogic-s9xxx-openwrt/actions/runs/10483811454/job/29037134348#step:14:9)xxx-openwrt/main/make-openwrt/kernel/stable/6.6.y.tar.gz: Cannot open: No such file or directory
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2
tar: Error is not recoverable: exiting now
 [💔] [ 6.6.y ] kernel decompression failed.
Error: Process completed with exit code 1.`

ophub, 2024-08-21:
我限了，你可以在model里加。

aguoxing, 2024-08-22:
> 我限了，你可以在model里加。 或者安装现在的6.6.y后自己armbian-update -k 6.1

是将model_database.conf中对应的设备:stable/6.6.y值修改为:stable/6.1.100或者:stable/6.1.y吗 改了还是没效果啊 勾了Auto use the latest kernel依旧下最新的 不勾就报内核下载失败

ophub, 2024-08-22:
fork后把自己仓库的model里对应的设备改为stable/6.1.y或stable/all

actions里面把use那行也指向你的仓库

closed 2025-06-02 by ophub