ophub/amlogic-s9xxx-openwrt #622: boot分区没有被正常挂载

jsdfhasuh, 2024-08-18:
**Device Information | 设备信息**
 - SOC: AM6B-plus
 - Model [e.g. HK1]

**OpenWrt Version | 系统版本**
 - Kernel Version: 5.15.100]
 - Release: official

**Describe the bug | 问题描述**
启动之后发现boot分区没有被正常挂载

[image]

jsdfhasuh, 2024-08-18:
[test.txt](https://github.com/user-attachments/files/16648128/test.txt)
openwrt official 编译选项

ophub, 2024-08-18:
[image]

[image]

[image]

ophub, 2024-08-18:
[image]

[image]

jsdfhasuh, 2024-08-18:
[Uploading blkid.log…]()

jsdfhasuh, 2024-08-18:
[blkid.log](https://github.com/user-attachments/files/16648803/blkid.log)

jsdfhasuh, 2024-08-18:
[dmesg.log](https://github.com/user-attachments/files/16648813/dmesg.log)

jsdfhasuh, 2024-08-18:
我镜像里面没有lsblk，我现在改一下

jsdfhasuh, 2024-08-18:
改了也没用，好像是mount 不支持label属性
mount: mounting LABEL="BOOT" on /boot failed: No such file or directory

closed 2024-08-31 by ophub