ophub/amlogic-s9xxx-openwrt #623: 安装 OpenWrt提示成功后无法启动

Jetlung, 2024-08-21:
**Device Information | 设备信息**
 - SOC: [e.g. S905L3A]
 - Model [e.g. M403A]

**OpenWrt Version | 系统版本**
 - Kernel Version: [k6.6.43]
 - Release: [openwrt_lede_amlogic_s905l3a_k6.6.43_2024.08.01.img.gz]

**Describe the bug | 问题描述**
安装 OpenWrt成功后无法启动，尝试过同版本前后几天的版次，同样情况

Please provide a detailed description of the issue and, if necessary, attach a screenshot.
使用 Rufus 或者 balenaEtcher写入U盘，按照安装 OpenWrt 的相关说明，点击 安装 OpenWrt 按钮进行安装（第2步）显示成功后，拔掉U盘无法启动，停留在花屏界面；同样的工具设备写入ARMBIAN是能正常启动并设置。

ophub, 2024-08-21:
从首页的设备列表点进去，查找你的设备，进之前的帖子里可以看到安卓系统的下载地址，先安装对应的安卓。或者到这里找：https://github.com/ophub/kernel/releases/tag/tools

farmerii, 2024-08-22:
Device Information | 设备信息
SOC: [e.g. S905L3]
Model [e.g. B8603.2AV-U]
OpenWrt Version | 系统版本
Kernel Version: [k6.6.43]
Release: [openwrt_official_amlogic_s905l3-cm211_k6.6.43_2024.08.01.img]

我也遇到了同样的问题，下载的cm211-1的包可以启动openwrt。但是用晶晨宝盒安装到emmc就不能启动了，同样是选择的cm211-1。
Current device: Amlogic Meson GXL (S905L3b) MBH-M302A Box [ amlogic ]选了几个包都是这个提示，是不是选设备型号的时候有bug
刷armbian选择第121都可以正常安装到emmc中，openwrt就不行。

ophub, 2024-08-23:
不断有新设备发现并加入model列表，所以设备序列号会被不断调整，你在仓库看到的121和你下载的固件里的121不一定一样。所以不要记这个序号，每次看自己下载的固件里安装列表里的实际选项。

Jetlung, 2024-08-24:
过程中都显示成功，这次连花屏都没有了，盒子显示绿灯，ping不通

farmerii, 2024-08-24:
> 不断有新设备发现并加入model列表，所以设备序列号会被不断调整，你在仓库看到的121和你下载的固件里的121不一定一样。所以不要记这个序号，每次看自己下载的固件里安装列表里的实际选项。

可能我表述的不是很清楚吧，我确实是按设备型号选的。提示安装成功之后死活登录不了web界面，按我上面的方法自定义安转才行

farmerii, 2024-08-24:
> 过程中都显示成功，这次连花屏都没有了，盒子显示绿灯，ping不通

我看有人说只有905l3a的芯片才有hdmi输出，其他的没有，反正我这款盒子是全黑屏的。
另外你按我的方法试试，或者你修改了默认的ip吗？默认lan口是固定ip，不是dhcp哦

Jetlung, 2024-08-24:
> > 过程中都显示成功，这次连花屏都没有了，盒子显示绿灯，ping不通
> 
> 我看有人说只有905l3a的芯片才有hdmi输出，其他的没有，反正我这款盒子是全黑屏的。 另外你按我的方法试试，或者你修改了默认的ip吗？默认lan口是固定ip，不是dhcp哦

巧了，我的就是S905l3a，没有更改IP，可是就连接不上；迟点我再试试你说的方法，感谢

ophub, 2024-08-24:
你换刷过贡献者分享的安卓系统了吗？要先换安卓，再安装armbian

farmerii, 2024-08-24:
> 你换刷过贡献者分享的安卓系统了吗？要先换安卓，再安装armbian

我的盒子型号非常冷门，刷的不是这里下载的安卓包，而且我买了刷机神器，不用拆机，所以刷起来跟简单。

ophub, 2024-08-24:
很多魔百和在使用armbian/openwrt系统时没有HDMI输出，插电开机1分钟后，在浏览器敲默认ip：192.168.1.1试试，自己的电脑也要设置到相同网段里才能访问到。

Jetlung, 2024-08-24:
> 你换刷过贡献者分享的安卓系统了吗？要先换安卓，再安装armbian

谢谢博主回复；我刷armbian是成功的，能正常使用；刷openwrt我都是照着网上的视频操作，也像另外一位同志那样购买了刷机神器，下方为我测试的流程；

1、插入刷机神器，接到电脑USB，使用USB_Burning_Tool刷android_tv_m401a_s905l3a.tar解压出来的img，成功；
2、balenaEtcher或rufus-4.4_x86_for-windows(两个都试过)把openwrt_lede_amlogic_s905l3a_k6.6.43_2024.08.01.img写入U盘，并启动DiskGenius扩容分区，修改uEnv的"meson-g12a-s905l3a-m401a.dtb"，退出U盘；
3、开心电视助手4.0连接后选择调试-晶晨芯片-从U盘/SD卡启动；
4、代码跑完后在浏览器输入192.168.1.1；点击系统-晶晨宝盒-安装openwrt-选择设备型号[306]M401A,UNT403A,B863AV3.2-M(也试过序号[120])，直至出现Successful installed...，重启让U盘带跑一次，拔掉U盘，再次启动就花屏或者无显示了。

Jetlung, 2024-08-24:
> 很多魔百和在使用armbian/openwrt系统时没有HDMI输出，插电开机1分钟后，在浏览器敲默认ip：192.168.1.1试试，自己的电脑也要设置到相同网段里才能访问到。

我刷好armbian后是能在HDMI输出的；刷好openwrt显示器无显等了10分钟，路由器是192.168.1.3，电脑192.168.1.44，在浏览器输入192.168.1.1是没有反应的。

Jetlung, 2024-08-24:
> 解决了，我重新将U盘写入armbian，运行armbian-install，然后抄下第121的选项配置。 再在安装openwrt的时候手动自动义dtb文件名，填入armbian中对应的信息就行了。 [ OPTIONS ] Please Input ID: 121 [ INFO ] Input Box ID: [ 121 ] [ INFO ] Model Name: [ CM211-1,M411A,B860AV3.2M,B860AV2.1U ] [ INFO ] FDTFILE: [ meson-gxl-s905l3b-m302a.dtb ] [ INFO ] MAINLINE_UBOOT: [ ] [ INFO ] BOOTLOADER_IMG: [ ] [ INFO ] UBOOT_OVERLOAD: [ u-boot-p212.bin ] [ INFO ] NEED_OVERLOAD: [ no ]

这个方法试了，没有改变-~-

farmerii, 2024-08-24:
> 谢谢博主回复；我刷armbian是成功的，能正常使用；刷openwrt我都是照着网上的视频操作，也像另外一位同志那样购买了刷机神器，下方为我测试的流程；
>
> 1、插入刷机神器，接到电脑USB，使用USB_Burning_Tool刷android_tv_m401a_s905l3a.tar解压出来的img，成功； [...]

你没仔细看我说的啊，在安装armbian到emmc的时候有一步会提示这次用的是那个dtb和u-boot，你安装armbian到emmc如果可以用的话，在安装openwrt的时候选自定义安装，再填进dtb、cpu和uboot就行

Jetlung, 2024-08-24:
不是这个吗？我的是[306]啊
dtb: meson-g12a-s905l3a-m401a.dtb
soc名称是啥？我随便填了个
uboot: u-boot-s905.bin
[image]

farmerii, 2024-08-24:
soc就是s905l3a啊

Jetlung, 2024-08-24:
我的是[306]；我刚才试过你说的方法，在晶晨盒子选择设备型号的时候自定义，我刚才描述的是我之前按照视频正常的流程，不知道哪步出错
dtb: meson-g12a-s905l3a-m401a.dtb
soc: S905l3a
uboot: u-boot-s905.bin
[image]

Jetlung, 2024-08-24:
> 你没仔细看我说的啊，在安装armbian到emmc的时候有一步会提示这次用的是那个dtb和u-boot，你安装armbian到emmc如果可以用的话，在安装openwrt的时候选自定义安装，再填进dtb、cpu和uboot就行

按照你说的那个方式，我又完完整整的操作一次，提示成功；我觉得跟这个关联没有那么大，如果选择的设备型号不符的话，是写入不成功的，博主ophub应该有考虑这个问题
[image]

ophub, 2024-08-24:
换6.1.y的固件，不要试6.6.y

farmerii, 2024-08-25:
> > > > 你换刷过贡献者分享的安卓系统了吗？要先换安卓，再安装armbian
> > > 
> > > 
> > > 谢谢博主回复；我刷armbian是成功的，能正常使用；刷openwrt我都是照着网上的视频操作，也像另外一位同志那样购买了刷机神器，下方为我测试的流程；
> > > 1、插入刷机神器，接到电脑USB，使用USB_Burning_Tool刷android_tv_m401a_s905l3a.tar解压出来的img，成功； 2、balenaEtcher或rufus-4.4_x86_for-windows(两个都试过)把openwrt_lede_amlogic_s905l3a_k6.6.43_2024.08.01.img写入U盘，并启动DiskGenius扩容分区，修改uEnv的"meson-g12a-s905l3a-m401a.dtb"，退出U盘； 3、开心电视助手4.0连接后选择调试-晶晨芯片-从U盘/SD卡启动； 4、代码跑完后在浏览器输入192.168.1.1；点击系统-晶晨宝盒-安装openwrt-选择设备型号[306]M401A,UNT403A,B863AV3.2-M(也试过序号[120])，直至出现Successful installed...，重启让U盘带跑一次，拔掉U盘，再次启动就花屏或者无显示了。
> > 
> > 
> > 你没仔细看我说的啊，在安装armbian到emmc的时候有一步会提示这次用的是那个dtb和u-boot，你安装armbian到emmc如果可以用的话，在安装openwrt的时候选自定义安装，再填进dtb、cpu和uboot就行
> 
> 按照你说的那个方式，我又完完整整的操作一次，提示成功；我觉得跟这个关联没有那么大，如果选择的设备型号不符的话，是写入不成功的，博主ophub应该有考虑这个问题 ![openwrt](https://private-user-images.githubusercontent.com/176786013/361150705-c0066ff9-455d-49fd-ac0f-7e29d470f6fa.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ1NDU4NDgsIm5iZiI6MTcyNDU0NTU0OCwicGF0aCI6Ii8xNzY3ODYwMTMvMzYxMTUwNzA1LWMwMDY2ZmY5LTQ1NWQtNDlmZC1hYzBmLTdlMjlkNDcwZjZmYS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODI1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgyNVQwMDI1NDhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hOGFiOGYzZDRkYzFkNWRjMDhhNDJiYzU0NDU5YzFmZGVlMDNhMGFkNzExZTNiYTQzMGM3OGZhZDE1NTFhNWQ1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.TafDSLs-2YEAYiB1DcHPkxXtLnLjTa9Za9co8uR1UL0)

看另一个issue我估计是跟uboot有关。也可能你的型号特殊，无法启动的问题跟我这盒子不一样。我看作者说6.0之前的版本启动跟uboot无关，可以试试。或者按作者说的试试6.1版本的。

Jetlung, 2024-08-25:
> 换6.1.y的固件，不要试6.6.y

非常感谢两位大佬的热心解答，换6.1.y的固件问题已解决，我尝试过很多个版本，唯独没有试过降低版本；同时也测试了安装openwrt-选择设备型号与自定义，两种方式均可行，再次感谢。

ophub, 2024-08-25:
我最近也发现6.6.y最近的内核有问题。

closed 2025-06-02 by ophub