ophub/amlogic-s9xxx-openwrt #624: boot partiotion size issue

drsamir21, 2024-08-24:
while using openwrt-update-amlogic command in terminal boot partion can't accomodate all files in amlogic s905x2.

ophub, 2024-08-31:
Take a screenshot and send your partition status: `lsblk`

drsamir21, 2024-08-31:
[image]

drsamir21, 2024-08-31:
I can install to emmc only after deleting some extra files from boot partition of sdcard.

ophub, 2024-09-01:
It seems that the installation script did not allocate enough space for the boot partition. Your approach is correct - you can delete the unnecessary u-boot-xxx.bin files from the boot partition.

ophub, 2024-09-01:
https://github.com/ophub/amlogic-s9xxx-openwrt/commit/95ea6c901e6fe44e708c86e504eb17dcc2bcf086

I adjusted the number of u-boot copies on the amlogic device so that 160MB of space will be sufficient.

drsamir21, 2024-09-02:
[image]

ophub, 2024-09-02:
https://github.com/ophub/amlogic-s9xxx-openwrt/commit/4a1af0a9df176ca4dc56691f21fda774c4e45885

Thank you for reporting the bug. I have just fixed the issue, and the new firmware is currently being compiled. It will be available in one hour.