ophub/amlogic-s9xxx-openwrt #633: Openmptcprouter on amlogic

DivineRuins, 2024-09-17:
First off all thank you for making his happen, openwrt working on my amlogic box is great, but i wanna take it a step further there is this project that is based on openwrt called openmptcprouter on github which uses a modified kernel with MPTCP enabled and custom openwrt packages to route traffic over different interfaces using shadowsocks and v2ray.. i want to build openmptcprouter but using your amlogic enabled kernel and patch it... i tried several things but failed

any quick guide on what should i do to compile openmptcprouter on amlogic s905, would really appreciate any help since am but lost in language and repositories and packers

Thank you!

ophub, 2024-09-18:
I don't want to discuss sensitive topics here. In some countries and regions, this is regulated content. In order to avoid trouble, I will delete your question after 1 hour.

DivineRuins, 2024-09-18:
Didnt know that i will edit the question

ophub, 2024-09-18:
https://github.com/ophub/kernel

https://github.com/ophub/amlogic-s9xxx-armbian/tree/main/compile-kernel