ophub/amlogic-s9xxx-openwrt #636: 使用openwrt-swap命令创建swap无法挂载

yuan6975, 2024-09-27:
**Device Information | 设备信息**
 - SOC: [e.g. S095l3a]
 - Model [e.g. M401A]

**OpenWrt Version | 系统版本**
 - Kernel Version: [e.g. 6.6.43]
 - Release: [e.g. official/lede/immortalwrt]

**Describe the bug | 问题描述**
All issues will only remain open for one week to prioritize resolving them.
After that period, they will be closed but can still continue to be discussed in the thread.
所有的问题都只保留一周的开启状态供重点关注解决，超时将关闭，但仍然可以长期继续跟帖讨论。

Please provide a detailed description of the issue and, if necessary, attach a screenshot.
详细描述问题，并在必要时附上屏幕截图。

下载的是本项目的openwrt-swap文件创建 btrfs文件系统 显示创建成功但是挂载还是失败

yuan6975, 2024-09-27:
[image]

yuan6975, 2024-09-27:
手动挂载swapon swapfile 显示
root@M401A-120G:/mnt/mmcblk2p4/.swap# swapon swapfile 
block: failed to swapon swapfile (-1)

ophub, 2024-09-28:
https://github.com/ophub/amlogic-s9xxx-openwrt/commit/beb43b0bbecc613eac8842996b78bf0704fadd0f

我刚才调整了swap的创建方法，你把这个更新里的2个文件覆盖本地的文件即可。
如果p4分区是ext4格式，之前的也是可以的，你创建无法使用，我猜你把p4分区格式化成了Btrfs格式了。

在 OpenWrt 系统里，Btrfs 格式的磁盘分区本身 不能直接在其上创建 swap 文件。
这是因为 Btrfs 文件系统不支持像 ext4 那样直接在已有的文件上启用 swap。

Btrfs 文件系统不支持 swap 文件的原因：
写时复制 (Copy-on-Write): Btrfs 是一种写时复制文件系统，这意味着对文件的任何修改都会创建一个新的数据块，而不是直接修改原始数据块。这对于数据完整性和快照功能非常有用，但对于 swap 文件来说效率很低，因为 swap 空间需要频繁地进行读写操作。
碎片化: Btrfs 的写时复制机制可能会导致 swap 文件的碎片化，从而降低 swap 性能。

解决方法：

尽管不能直接在 Btrfs 分区上创建 swap 文件，但你可以使用以下方法在 OpenWrt 上创建 swap 空间：
使用 loop 设备： 这是最常见的解决方法，创建一个虚拟的 loop 设备并将其格式化为 swap。
使用 U 盘或 SD 卡： 将外部存储设备格式化为 swap 分区，并挂载到 OpenWrt。
使用 zram： 利用内存压缩技术创建 swap 空间，适用于内存较大的设备。

yuan6975, 2024-09-28:
> [beb43b0](https://github.com/ophub/amlogic-s9xxx-openwrt/commit/beb43b0bbecc613eac8842996b78bf0704fadd0f)
> 
> 我刚才调整了swap的创建方法，你把这个更新里的2个文件（/usr/sbin/openwrt-swap 和 /etc/custom_service/start_service.sh）覆盖本地的文件即可。 如果p4分区是ext4格式，之前的也是可以的，你创建无法使用，我猜你把p4分区格式化成了Btrfs格式了。
> 
> 在 OpenWrt 系统里，Btrfs 格式的磁盘分区本身 不能直接在其上创建 swap 文件。 这是因为 Btrfs 文件系统不支持像 ext4 那样直接在已有的文件上启用 swap。
> 
> Btrfs 文件系统不支持 swap 文件的原因： 写时复制 (Copy-on-Write): Btrfs 是一种写时复制文件系统，这意味着对文件的任何修改都会创建一个新的数据块，而不是直接修改原始数据块。这对于数据完整性和快照功能非常有用，但对于 swap 文件来说效率很低，因为 swap 空间需要频繁地进行读写操作。 碎片化: Btrfs 的写时复制机制可能会导致 swap 文件的碎片化，从而降低 swap 性能。
> 
> 解决方法：
> 
> 尽管不能直接在 Btrfs 分区上创建 swap 文件，但你可以使用以下方法在 OpenWrt 上创建 swap 空间： 使用 loop 设备： 这是最常见的解决方法，创建一个虚拟的 loop 设备并将其格式化为 swap。 使用 U 盘或 SD 卡： 将外部存储设备格式化为 swap 分区，并挂载到 OpenWrt。 使用 zram： 利用内存压缩技术创建 swap 空间，适用于内存较大的设备。
> 
> ![Snip20240928_3](https://private-user-images.githubusercontent.com/68696949/371727328-55dac3a2-9a10-4d9e-9387-edf01bc7790d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc1MDMxODgsIm5iZiI6MTcyNzUwMjg4OCwicGF0aCI6Ii82ODY5Njk0OS8zNzE3MjczMjgtNTVkYWMzYTItOWExMC00ZDllLTkzODctZWRmMDFiYzc3OTBkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI4VDA1NTQ0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZiNzcxNDUxOGU2MzUyNGJhYjExODc4NGY3OWYwNzBlZWU3MWFhNWYwNWJmMGI1NGQzNjBhMjU0MzkzYjdmZGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.GGgkZjlwhu3CoENmVBHCbeVdhl30y7Ixs3rHs0uLVcI)
> 
> ![Snip20240928_4](https://private-user-images.githubusercontent.com/68696949/371727331-ed88c444-d6ac-4504-aca2-380f8efbfa82.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc1MDMxODgsIm5iZiI6MTcyNzUwMjg4OCwicGF0aCI6Ii82ODY5Njk0OS8zNzE3MjczMzEtZWQ4OGM0NDQtZDZhYy00NTA0LWFjYTItMzgwZjhlZmJmYTgyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTI4VDA1NTQ0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFiMmEwZTgzN2ExNTVmZjliMmY3ODlhYzM2Mzk1OGVlMGI1MjY2NzRmMTFjNTJkODNiYjA4NzY4ZjE2ZDRlZGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.2zQM7_cP2Trq4lfFZ_rIHERPddJovNVk3onmS00BYgM)

好的谢谢  如果我想把p4改成ext4 需要重新刷完整包吗? 还是说卸载p4 在磁盘管理修改/dev/mmblk2 格式化成ext4  因为我最开始5.15.*内核是btrfs文件系统升级的6.6.43

ophub, 2024-09-28:
换p4格式要格式化分区，数据备份了格式化。

yuan6975, 2024-09-28:
> 换p4格式要格式化分区，数据备份了格式化。

只需要格式化p4就可以了吧  p2 p3 还是可以保持btrfs对吧

ophub, 2024-09-28:
对，只卸载p4，格式化

ophub, 2024-09-28:
现在的脚本不用换ext4了，直接下载最新脚本覆盖本地的，btrfs也可以使用swap

yuan6975, 2024-09-28:
> 现在的脚本不用换ext4了，直接下载最新脚本覆盖本地的，btrfs也可以使用swap

好的  我有个小盒子只有1g内存 创建了1g swap 跑pcdn的时候p4分区读取速度10M/S htop上面看cpu 红色占80%是内核进程 是不是swap的原因

ophub, 2024-09-28:
swap是一个独立文件，而不是整个分区，所以分区的读写速度和swap无关

yuan6975, 2024-10-01:
> swap是一个独立文件，而不是整个分区，所以分区的读写速度和swap无关

还有一个问题就是开机不能自动挂载  挂载点添加/mnt/mmcblk2p4/.swap/swapfile不能启动 开机应该怎么自启

ophub, 2024-10-01:
看我上面发的信息，我更新了两个文件，其中 start_service.sh 是负责处理启动文件的，把这个文件也覆盖更新了。

yuan6975, 2024-10-01:
> 看我上面发的信息，我更新了两个文件，其中 start_service.sh 是负责处理启动文件的，把这个文件也覆盖更新了。

把start_service.sh添加到开机启动里面就可以了吧  手动./start_service.sh就启动swap了

yuan6975, 2024-10-01:
> 在/etc目录下创建custom_service目录，下载完整的start_service.sh文件放在这个目录里，分别执行下面两个命令，添加执行权限和开机自启动：
> 
> ```
> chmod +x /etc/custom_service/start_service.sh
> 
> sed -i '/^exit 0/i\bash /etc/custom_service/start_service.sh' /etc/rc.local
> ```
> 
> 这样就可以了。

好的 谢谢

closed 2025-02-10 by ophub