ophub/amlogic-s9xxx-openwrt #648: Custom kernel not working

Manssizz, 2024-10-26:
**Device Information | 设备信息**
 - SOC: All
 - Model All

**OpenWrt Version | 系统版本**
 - Kernel Version: 5.15.y 6.1.y 6.6y
 - Release: All

**Describe the bug | 问题描述**
I have custom kernel with low latency tags in my [kernel repo](https://github.com/Manssizz/ophub-kernel). I was add my own kernel repo and kernel usage value. Compiled work but still used ophub kernel. How to use custom kernel? i've try change some value in remake script

Please provide a detailed description of the issue and, if necessary, attach a screenshot.
[image]
[image]

ophub, 2024-10-27:
Tags use name:
kernel_stable
kernel_beta
kernel_dev

Manssizz, 2024-10-27:
so... only support 3 tags (stable, beta,dev)?

Manssizz, 2024-10-29:
ok. thanks. closed

closed 2024-10-29 by Manssizz