ophub/amlogic-s9xxx-openwrt #650: 请求加入对mihomo, homeproxy的内核支持

sztuxp, 2024-10-29:
大佬，希望固件能加入mihomo, homeproxy的内核支持

**Device Information | 设备信息**
 - SOC: N1
 - Model [N1]

**OpenWrt Version | 系统版本**
 - Kernel Version: 6.6.58
 - Release: [official ]

**Describe the bug | 问题描述**
N1的盒子，下载的6.6.57，然后升级内核到6.6.58
安装mihomo, 显示下面的错误信息
Installing mihomo (2024.10.20~3e966e82-r2) to root...
Downloading https://morytyann.github.io/OpenWrt-mihomo/aarch64_generic/mihomo/mihomo_2024.10.20~3e966e82-r2_aarch64_generic.ipk
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-nfnetlink
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-nf-reject
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-nf-reject6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-nf-conntrack
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-nf-conntrack6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-nf-nat
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-nf-log
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-nf-log6
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-crypto-hash
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-crypto-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-lib-crc32c
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-nft-core
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-nft-fib
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-nf-flow
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-nft-nat
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-nft-offload
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-inet-diag
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-nf-tproxy
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-nft-tproxy
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nft-tproxy found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1) for kmod-tun
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for mihomo:
 * 	kernel (= 6.6.58~ecb9bbd438ff5b5a1a45969ca247bf60-r1)
 * opkg_install_cmd: Cannot install package mihomo.

sztuxp, 2024-10-29:
补充一下，[OpenWrt-mihomo](https://github.com/morytyann/OpenWrt-mihomo)  
需求：OpenWrt >= 22.03  Linux Kernel >= 5.10 firewall4

ophub, 2024-10-29:
敏感插件不讨论，你可以根据仓库的教程自己添加个性化软件包