ophub/amlogic-s9xxx-openwrt #652: 请教大佬几个编译问题

sztuxp, 2024-10-30:
首先感谢大佬提供的好项目，小白也能成功定制编译了。
我是fork项目最新代码，用action进行编译，目前几个问题
1、设备，S905D，选择编译openwrt-man，定制修改，都成功编译。但无论选择是save, 还是temp，都会发布到releases地方，本来理解tmp是放在actions处。
2、选择immortalwrt-master, 编译出错，exit code 2

ophub, 2024-11-01:
这个Actions示例里，把上传到Actions里的代码删除了。因为这里需要登录才能下载，你去官方或者其他仓库看看这个官方上传的Actions。

放在releases里不需要登录也可以下载，方便大家下载使用。

sztuxp, 2024-11-01:
明白了，谢谢大佬回复！第 二个问题，因为openwrt不能关闭Dnsmasq的DNS劫持功能，immortalwrt是可以，现在等大佬出一个immortalwrt的action可编译版本，在那个基础上定制一下固件。

ophub, 2024-11-01:
现在就可以。

[image]

sztuxp, 2024-11-01:
大佬，我现在就是用这个action 选 immortalwrt-master编译不过，选 openwrt-main就可以，不知道是什么原因?

sztuxp, 2024-11-01:
不好意思，用一个小号重新fork，编译成功了。谢谢!

ophub, 2024-11-01:
好的。默认的config只选择了少量几个插件，一般来说都能正常编译通过，偶遇失败可以到上游的源码仓库看看他的Issue里是不是有人反馈和你日志里一样的编译错误，查看解决办法或者等上游修复后再编译。如果碰到上游有大版本更新，往往需要重新本地选下config生成新的配置文件。

closed 2025-02-10 by ophub