ophub/amlogic-s9xxx-openwrt #660: 上游源全部采用apk方式，软件包无法更新，请问如何解决

sztuxp, 2024-11-17:
上游源全部采用apk方式，软件包无法更新，请问如何解决

jlhhb, 2024-11-17:
同问：系统信息如下：
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='R2024.11.01'
DISTRIB_TARGET='armsr/armv8'
DISTRIB_ARCH='aarch64_generic'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r0-2b173ab'
DISTRIB_TAINTS='no-all'
DISTRIB_SOURCECODE='official'

S905L  Q5盒子，使用自带源的时候更新软件提示：
Downloading http://downloads.openwrt.org/snapshots/targets/armsr/armv8/packages/Packages.gz
*** Failed to download the package list from http://downloads.openwrt.org/snapshots/targets/armsr/armv8/packages/Packages.gz

Downloading http://downloads.openwrt.org/snapshots/packages/aarch64_generic/base/Packages.gz
*** Failed to download the package list from http://downloads.openwrt.org/snapshots/packages/aarch64_generic/base/Packages.gz

Downloading http://downloads.openwrt.org/snapshots/packages/aarch64_generic/luci/Packages.gz
*** Failed to download the package list from http://downloads.openwrt.org/snapshots/packages/aarch64_generic/luci/Packages.gz

Downloading http://downloads.openwrt.org/snapshots/packages/aarch64_generic/packages/Packages.gz
*** Failed to download the package list from http://downloads.openwrt.org/snapshots/packages/aarch64_generic/packages/Packages.gz

Downloading http://downloads.openwrt.org/snapshots/packages/aarch64_generic/routing/Packages.gz
*** Failed to download the package list from http://downloads.openwrt.org/snapshots/packages/aarch64_generic/routing/Packages.gz

Downloading http://downloads.openwrt.org/snapshots/packages/aarch64_generic/telephony/Packages.gz
*** Failed to download the package list from http://downloads.openwrt.org/snapshots/packages/aarch64_generic/telephony/Packages.gz

Collected errors:
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/targets/armsr/armv8/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/aarch64_generic/base/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/aarch64_generic/luci/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/aarch64_generic/packages/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/aarch64_generic/routing/Packages.gz, wget returned 8.
 * opkg_download: Failed to download http://downloads.openwrt.org/snapshots/packages/aarch64_generic/telephony/Packages.gz, wget returned 8.

检测wget正常，排除网络，DNS，wget问题

jlhhb, 2024-11-17:
补充：
root@OpenWrt:/tmp# wget https://mirror-03.infra.openwrt.org/snapshots/packages/aarch64_generic/packages/zerotier-1.14.1-r1.apk
--2024-11-17 19:55:17--  https://mirror-03.infra.openwrt.org/snapshots/packages/aarch64_generic/packages/zerotier-1.14.1-r1.apk
Resolving mirror-03.infra.openwrt.org... 198.18.1.26
Connecting to mirror-03.infra.openwrt.org|198.18.1.26|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 554258 (541K) [application/octet-stream]
Saving to: 'zerotier-1.14.1-r1.apk'

zerotier-1.14.1-r1.apk              100%[================================================================>] 541.27K   276KB/s    in 2.0s

2024-11-17 19:55:21 (276 KB/s) - 'zerotier-1.14.1-r1.apk' saved [554258/554258]

root@OpenWrt:/tmp# chmod 777 zerotier-1.14.1-r1.apk
root@OpenWrt:/tmp# opkg install zerotier-1.14.1-r1.apk
Unknown package 'zerotier-1.14.1-r1.apk'.
Collected errors:
 * opkg_install_cmd: Cannot install package zerotier-1.14.1-r1.apk.

jlhhb, 2024-11-17:
再补充：试过official、immortalwrt，版本均有以上问题

xrgzs, 2024-11-22:
同问，apk是趋势，能不能给官方的构建改成apk？

ophub, 2024-11-22:
我2天前刚编译了一遍，official、immortalwrt的2个tags都使用的他们的源码，编译的不是最新的apk固件？你知道怎么改编译配置项支持这个？

sztuxp, 2024-11-22:
> 看官方的帖子说main分支已经是apk了（11月16日发的），我2天前刚编译了一遍，official、immortalwrt的2个tags都使用的他们的源码，编译的不是最新的apk固件？
> 
> 使用Image Builder是调用官方编译的固件进行重构，你使用这个编译固件总该是apk了吧。
> 
> https://forum.openwrt.org/t/the-future-is-now-opkg-vs-apk/201164
> 
> 支持apk可能是需要打开这个选项。
> 
> https://github.com/ophub/amlogic-s9xxx-openwrt/blob/5007867d3e8b4952de624c4ae8b48910571e835e/config/openwrt-main/config#L191

大佬，我特意下了你前2天编译的版本，软件包更新列表还是错误，请问一下，APK软件包有新的更新列表格式吗？

ophub, 2024-11-22:
你把网络接口配置好了可以正常访问网络了吗？这个源是官方自带的，理论上网络正常就可以下载，或者这个源站在国外？你要用那种网？

sztuxp, 2024-11-24:
网络接口没问题，个人感觉APK列出源站列表时，不需要Packages.gz ，因此新的源站都没有这个文件，但用旧的格式去更新列表时，因找不到Packages.gz类似的文件，就会出错。请看二楼报错信息就知道。

Madao-MAX, 2024-12-07:
> 网络接口没问题，个人感觉APK列出源站列表时，不需要Packages.gz ，因此新的源站都没有这个文件，但用旧的格式去更新列表时，因找不到Packages.gz类似的索引文件，就会出错, 新的软件库好像是一个json索引文件，比如 [https://downloads.openwrt.org/snapshots/packages/aarch64_generic/routing/index.json，而不是http://downloads.openwrt.org/snapshots/packages/aarch64_generic/routing/Packages.gz](https://downloads.openwrt.org/snapshots/packages/aarch64_generic/routing/index.json%EF%BC%8C%E8%80%8C%E4%B8%8D%E6%98%AFhttp://downloads.openwrt.org/snapshots/packages/aarch64_generic/routing/Packages.gz) 。请看二楼报错信息就知道。

改成这个源试试mirrors.cloud.tencent.com

lpfxh, 2024-12-28:
您好是因为最新的安装包还是ipk的安装方式吗

davidwang558, 2025-01-04:
ImmortalWrt 遇到一样的问题 我原本需要安装passwall的 没想到没有成功 然后发现也无法opkg update

ophub, 2025-01-05:
12月的最新版改成apk了

closed 2025-02-10 by ophub