ophub/amlogic-s9xxx-openwrt #663: 黑龙江新魔百盒M201-d S905b 芯片 无法写入emmc

andylovephysics, 2024-11-29:
**Device Information | 设备信息**
 - SOC: [e.g. S905L B]
1g+8g 
版型：
[image]
[image]

**OpenWrt Version | 系统版本**
 - Release: [https://github.com/ophub/amlogic-s9xxx-armbian/releases/download/Armbian_noble_save_2024.11/Armbian_24.11.0_amlogic_s905lb-r3300l_noble_6.6.62_server_2024.11.20.img.gz]

**Describe the bug | 问题描述**

我直接将release文件写入U盘，并使盒子从U盘更新系统。能正常进入armbian系统，但是在准备将系统写入emmc时，输入armbian-install时，显示如下错误：
```
root@armbian:~# armbian-install
[ STEPS ] Installing Armbian to internal eMMC...
[ STEPS ] Checking dependencies...
[ INFO ] Dependency check completed. Proceeding installation...
[ STEPS ] Initializing the environment...
[ INFO ] Use mainline u-boot: [ no ]
[ INFO ] Use ampart tool: [ yes ]
[ INFO ] Show all devices: [ no ]
[ ERROR ] Internal eMMC storage was't found in this device!
```
[image]

通过在网上查询教程尝试对FDT进行修改，默认值为meson-gxl-s905x-p212.dtb，修改为
FDT=/dtb/amlogic/meson-gxl-s905l3b-m302a.dtb

两个都无法进行写入emmc出现同样的错误，即
```
root@armbian:~# armbian-install
[ STEPS ] Installing Armbian to internal eMMC...
[ STEPS ] Checking dependencies...
[ INFO ] Dependency check completed. Proceeding installation...
[ STEPS ] Initializing the environment...
[ INFO ] Use mainline u-boot: [ no ]
[ INFO ] Use ampart tool: [ yes ]
[ INFO ] Show all devices: [ no ]
[ ERROR ] Internal eMMC storage was't found in this device!
```

请问该如何处理

andylovephysics, 2024-11-29:
P.S. 第一次刷入 Armbian_24.11.0_amlogic_s905l-mg101_noble_6.6.58_server_2024.11.01.img 时也能够正常进入系统，同样也无法写入emmc

ophub, 2024-11-29:
nand存储不能写入，emmc可以。
在终端里使用lsblk看看能不能看到emmc
如果确定是emmc而看不到，可以尝试把dtb的频率降低到25hz再试试，方法看仓库的文档里如何反编译dtb以及调整频率的值的方法。

andylovephysics, 2024-11-30:
感谢回复！lsblk能够看到mmc的储存，如图所示并且尝试更改了meson-gxl-s905l3b-m302a.dtb中的max-frequency。此处的max-frequency搜索到了3处，并将它们都改为了0x5f5e100经过更改和上传及编译为dtb后，仍然在执行armbian-install时报错，如图除此之外，我可能没有在仓库文档中找到调节频率有关的内容，如果有的话请麻烦给一下链接这次的降频操作是参照(https://post.m.smzdm.com/p/apmpnqx9/)[此链接]进行操作的请问这次该怎么办呢？

andylovephysics, 2024-11-30:
感谢回复！lsblk能够看到mmc的储存，如图所示
[image]

我感觉这个显示的是我外接的sd卡，本地的储存（8g）没有被显示。

然后尝试更改了meson-gxl-s905l3b-m302a.dtb中的max-frequency。
max-frequency搜索到了3处，并将它们都改为了0x5f5e100经过更改和上传及编译为dtb后，仍然在执行armbian-install时报错，如图

[image]

除此之外，我可能没有在仓库文档中找到调节频率有关的内容，如果有的话请麻烦给一下链接

这次的降频操作是参照https://post.m.smzdm.com/p/apmpnqx9 进行操作的请问这次该怎么办呢？

andylovephysics, 2024-11-30:
P.S. 这是板载的两个存储芯片的细节照，从这里能看出来什么吗

[image]
这个应该是nand（它能做运存吗？）

[image]
这个不确定

andylovephysics, 2024-11-30:
这是所有已加载节点的设备

[image]

closed 2025-02-10 by ophub