ophub/amlogic-s9xxx-openwrt #665: 有支持s905x4的计划吗？

theosoft-git, 2024-12-04:
支持下s905x4应该不难吧？

ophub, 2024-12-05:
不难，等kernel.org支持了就可以了

wuai1024, 2024-12-17:
@ophub 大佬，赚3 Pro（amlogic-s912），希望能支持下

wuai1024, 2024-12-17:
[image]

wuai1024, 2024-12-17:
> ![image](https://private-user-images.githubusercontent.com/13379708/396549104-1797b4de-26c3-4650-a64c-0b31a8da14b2.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ0NDgzMDEsIm5iZiI6MTczNDQ0ODAwMSwicGF0aCI6Ii8xMzM3OTcwOC8zOTY1NDkxMDQtMTc5N2I0ZGUtMjZjMy00NjUwLWE2NGMtMGIzMWE4ZGExNGIyLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIxN1QxNTA2NDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xNjU5ZmUxOTkwYTgzODIxZTUyODEzMjRlNjQ3ODc0YzFhZWM2YWJjMDMwYmNkNWVmOTliMTg4MDZjNjdiYWE0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.IssgzzjA2WfLYtqpI7UQox519J4LYqBkUpxpMNlSuXk)

这2个镜像，都是卡住在这里。
openwrt_official_amlogic_s912-m8s-pro_k6.1.119_2024.12.15.img
openwrt_official_amlogic_s912-m8s-pro_k6.6.65_2024.12.15.img

closed 2025-02-10 by ophub