ophub/amlogic-s9xxx-openwrt #671: 本地化打包错误，挂载失败,请问一下这是什么问题？

jreeybin, 2024-12-18:
[ INFO ] [ 1 ] lists of OpenWrt board: [ e20c ]
[ INFO ] Kernel Repo: [ ophub/kernel ], Kernel Usage: [  ] 

[ INFO ] Server space usage before starting to compile: 
文件系统       类型  大小  已用  可用 已用% 挂载点
/dev/sda2      ext4   79G   22G   53G   30% / 

[ STEPS ] Start making OpenWrt firmware...
(1.1) Start making OpenWrt [ e20c - rk35xx/5.10.160 ]. Remaining space is 52G.
 [🌿] (1/6) Make OpenWrt image.
 [🌿] (2/6) Extract OpenWrt files.
mount: /home/jreeybin/amlogic-s9xxx-openwrt/openwrt/tmp/5.10.160/e20c/rootfs: wrong fs type, bad option, bad superblock on /dev/loop21p2, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.
mount: /home/jreeybin/amlogic-s9xxx-openwrt/openwrt/tmp/5.10.160/e20c/rootfs: wrong fs type, bad option, bad superblock on /dev/loop21p2, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.
mount: /home/jreeybin/amlogic-s9xxx-openwrt/openwrt/tmp/5.10.160/e20c/rootfs: wrong fs type, bad option, bad superblock on /dev/loop21p2, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.
mount: /home/jreeybin/amlogic-s9xxx-openwrt/openwrt/tmp/5.10.160/e20c/rootfs: wrong fs type, bad option, bad superblock on /dev/loop21p2, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.
mount: /home/jreeybin/amlogic-s9xxx-openwrt/openwrt/tmp/5.10.160/e20c/rootfs: wrong fs type, bad option, bad superblock on /dev/loop21p2, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.
mount: /home/jreeybin/amlogic-s9xxx-openwrt/openwrt/tmp/5.10.160/e20c/rootfs: wrong fs type, bad option, bad superblock on /dev/loop21p2, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.
mount: /home/jreeybin/amlogic-s9xxx-openwrt/openwrt/tmp/5.10.160/e20c/rootfs: wrong fs type, bad option, bad superblock on /dev/loop21p2, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.
mount: /home/jreeybin/amlogic-s9xxx-openwrt/openwrt/tmp/5.10.160/e20c/rootfs: wrong fs type, bad option, bad superblock on /dev/loop21p2, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.
mount: /home/jreeybin/amlogic-s9xxx-openwrt/openwrt/tmp/5.10.160/e20c/rootfs: wrong fs type, bad option, bad superblock on /dev/loop21p2, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.
mount: /home/jreeybin/amlogic-s9xxx-openwrt/openwrt/tmp/5.10.160/e20c/rootfs: wrong fs type, bad option, bad superblock on /dev/loop21p2, missing codepage or helper program, or other error.
       dmesg(1) may have more information after failed mount system call.
 [💔] [ 11 ] attempts to mount failed.

ophub, 2024-12-19:
在什么系统上打包的呢

jreeybin, 2024-12-19:
在Ubuntu 24.04上面，试了两台机子都一样的结果，请问一下，在线编译哪里修改软件区为2048,我改了好多地方都没有改对，编译出来还是1024

ophub, 2024-12-19:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/f5422ab60d8d225f92229fc051bf77b3e9c1f188/remake#L153-L157

可能你的系统使用非ext4和btrfs格式，所以这个挂载命令会报错，你可以根据自己的磁盘类型修改。

`-s` 参数是修改rootrs区的大小的，不用改程序，首页有说明。

jreeybin, 2024-12-19:
好的，我再试试，我看了我的系统是ETX4的，rootrs分区我说是在GitHub Actions 进行编译时设置。

jreeybin, 2024-12-19:
虚拟机装的Ubuntu是不是不支持的

ophub, 2024-12-20:
我本地也是在虚拟机里装的ubuntu24.04测试正常。
在actions里使用-s需要自己加个选项，文本框输入对应的值。

jreeybin, 2024-12-20:
告诉我加到哪里吗，怎么加呢？麻烦了

ophub, 2024-12-20:
actions的yml里

jreeybin, 2024-12-21:
openwrt_size:
    description: "Set the rootfs size(Unit: MiB)."
    required: false
    default: "2048"
    是这样设置吗？

ophub, 2024-12-21:
还有下面的接收参数的地方

https://github.com/ophub/amlogic-s9xxx-openwrt/blob/f5422ab60d8d225f92229fc051bf77b3e9c1f188/.github/workflows/build-openwrt.yml#L390-L400

jreeybin, 2024-12-22:
设置了
openwrt_size:
description: "Set the rootfs size(Unit: MiB)."
required: false
default: "2048"
设置了
      - name: Packaging OpenWrt
        if: ${{ steps.clean.outputs.status }} == 'success' && !cancelled()
        uses: ophub/amlogic-s9xxx-openwrt@main
        with:
          openwrt_path: openwrt/output/*rootfs.tar.gz
          openwrt_board: ${{ inputs.openwrt_board }}
          openwrt_kernel: ${{ inputs.openwrt_kernel }}
          auto_kernel: ${{ inputs.auto_kernel }}
          kernel_repo: ${{ inputs.kernel_repo }}
          kernel_usage: ${{ inputs.kernel_usage }}
          builder_name: ${{ inputs.builder_name }}
          openwrt_size: ${{ inputs.openwrt_size }}
这样吗？打包出来还是1024的

ophub, 2024-12-22:
你刷USB启动看一下

jreeybin, 2024-12-23:
刷了，还是1024的

closed 2025-02-10 by ophub