ophub/amlogic-s9xxx-openwrt #672: fine本地打包指定kernel失败。

n14l, 2024-12-24:
本地打包fine3399在不指定kernel版本的时候可以正常打包，指定kernel版本会打包失败，提示下载错误，而且下载kernel的连接是http://....6.1.y。这是不是没有成功输入指定内核的命令？还是现在指定内核版本功能取消了？

ophub, 2024-12-24:
全部过程截图

ophub, 2024-12-24:
`sudo ./remake -b fine3399 -k 6.1.120 -a false`

n14l, 2024-12-24:
[image]
[image]
这是打包成功的

n14l, 2024-12-24:
[image]
[image]
这个是同样环境，只指定kernel版本的

iamyangyiok, 2025-01-31:
肯定是个编号 怎么会有字母

ophub, 2025-01-31:
感谢反馈错误，使用-a false指定固定版本内核的错误已经修复了，你拉取最新源码即可。

closed 2025-02-10 by ophub