ophub/amlogic-s9xxx-openwrt #685: 更新系统内核问题

mrsoshining, 2025-01-29:
大佬，如果更新系统内核或全系统，是还是之前安装的应用、相关配置还有docker的容器就都丢失了？因为SNAPSHOT内核一些软件没有集成、又通过包安装不上。谢谢

ophub, 2025-01-30:
更新内核什么也不会丢。
更新完整固件前，在插件设置里选择保留配置就不会丢。建议使用同一个仓库或者同一个来源的固件，新旧固件的插件相同就不会少了插件。

mrsoshining, 2025-01-30:
谢谢，因为有些软件包我怎么都安装不了，所以我想按照说明文档自己编译固件试一试。还有个问题，官方的固件版本是不能直接用在盒子上的吧，只能是fork大佬的仓库进行编译吧。

mrsoshining, 2025-01-30:
openwrt与immortalwrt互转应用、插件、设置就没了吧

mrsoshining, 2025-01-31:
我自己编译的固件应该还是有问题

ophub, 2025-02-01:
https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/.github/workflows/build-openwrt-with-imagebuilder.yml

用这个imagebuilder一会几分钟就编一个。是在官方固件的基础上添加插件，建议你看一下仓库的文档，自己编自己的固件。循序渐进地加，先加一个插件试试，对了再接着加，不要一下就加一堆，有问题不好排查是哪个的问题。另外有问题要尽量截图把问题发完整版，用语言表达有时候不清楚。

mrsoshining, 2025-02-01:
> https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/.github/workflows/build-openwrt-with-imagebuilder.yml
> 
> 用这个imagebuilder一会几分钟就编一个。是在官方固件的基础上添加插件，建议你看一下仓库的文档，自己编自己的固件。循序渐进地加，先加一个插件试试，对了再接着加，不要一下就加一堆，有问题不好排查是哪个的问题。另外有问题要尽量截图把问题发完整版，用语言表达有时候不清楚。

谢谢，我已经尝试用这个构建了，而且是在https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/config/immortalwrt-master/config增加的插件，我的操作方法对吗？

ophub, 2025-02-01:
对，openwrt官方和immortalwrt都支持imagebuilder，这两个都有他们预编译好的固件，可以直接在里面添加个性化插件。

mrsoshining, 2025-02-01:
谢谢，那位我升级卡在umount [ /dev/loop0p1 ] ...不动，usb重新安装也没反应，可能还是固件有问题、或者原来的系统有问题了，我也试过了openwrt-backup -g

ophub, 2025-02-01:
终端执行lsblk看看挂载点是什么情况

mrsoshining, 2025-02-01:
我记得执行过一次，看了没有loop0p1，但有个loop0p2

ophub, 2025-02-01:
p1是/boot
正常情况下是显示的，p1-p4都在才对

mrsoshining, 2025-02-01:
确实没什么基础、我再好好学习一下，谢谢了

mrsoshining, 2025-02-06:
> 谢谢，我已经尝试用这个构建了，而且是在[https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/config/immortalwrt-master/config增加的插件，我的操作方法对吗？](https://github.com/ophub/amlogic-s9xxx-openwrt/blob/main/config/immortalwr [...]

更新安装成功了，谢谢

closed 2025-02-10 by ophub