ophub/amlogic-s9xxx-openwrt #686: immortalwrt 23.05打包后fullcone 没有了

iamyangyiok, 2025-01-30:
**Device Information | 设备信息**
PANTHER X2

**OpenWrt Version | 系统版本**
 - Kernel Version: 6.1.127
 - Release: immortalwrt 23.05

 immortalwrt 23.05打包后fullcone 没有了。是新核心不支持吗？ 还是BUG

mrsoshining, 2025-01-30:
23.05需要自己编译吧，我自己弄了之后，升级时总卡在umount

ophub, 2025-01-30:
升级时总卡在umount可以先在终端下输入命令：`openwrt-backup -g`，重新整理下挂载点后再运行升级固件。

fullcone支持，你的原始rootfs编译了吗？你可以试试更新其他系列的内核看看有没有。

iamyangyiok, 2025-01-30:
> 升级时总卡在umount可以先在终端下输入命令：`openwrt-backup -g`，重新整理下挂载点后再运行升级固件。
> 
> fullcone支持，你的原始rootfs编译了吗？你可以试试更新其他系列的内核看看有没有。

是的自己immtoratal编译rootfs生成的，选中了kmod-nft-fullcone 及相关的模块，但是 打包后相应横块也都在，就是无法FULLCONE ，包括选项都没有， 我记得天灵说FULLCONE要核心支持，现在不确认问题点，我用23.05 23.05.4 生成都有这问题，不知道是不是nftables核心兼容的问题。

iamyangyiok, 2025-01-30:
> 升级时总卡在umount可以先在终端下输入命令：`openwrt-backup -g`，重新整理下挂载点后再运行升级固件。
> 
> fullcone支持，你的原始rootfs编译了吗？你可以试试更新其他系列的内核看看有没有。

是的自己immtoratal编译rootfs生成的，选中了kmod-nft-fullcone 及相关的模块，但是 打包后相应横块也都在，就是无法FULLCONE ，包括选项都没有， 我记得天灵说FULLCONE要核心支持，现在不确认问题点，我用23.05 23.05.4 生成都有这问题，不知道是不是nftables核心兼容的问题。 用stable 和flippy 的试过都不能开启fullcone 支持

> **Device Information | 设备信息** PANTHER X2
> 
> **OpenWrt Version | 系统版本**
> 
> * Kernel Version: 6.1.127
> * Release: immortalwrt 23.05
> 
> immortalwrt 23.05打包后fullcone 没有了。是新核心不支持吗？ 还是BUG

iamyangyiok, 2025-01-30:
不好意思 网卡了 连发了

iamyangyiok, 2025-02-10:
更新：今天一群友测试 打包TIV3315 也是打包后 FULLCONE没有了，用的FLIPPY的核心。

closed 2025-06-02 by ophub