ophub/amlogic-s9xxx-openwrt #693: 两次上电才能连上网络

kelvin-ye, 2025-02-10:
**Device Information | 设备信息**
 - SOC: rk3328
 - Model BeiKeYun

**OpenWrt Version | 系统版本**
 - Kernel Version: k6.6.74-happy-new-year
 - Release:  [openwrt_rk3328_beikeyun_R25.01.01_k6.6.74-happy-new-year.img.gz]

**Describe the bug | 问题描述**
All issues will only remain open for one week to prioritize resolving them.
After that period, they will be closed but can still continue to be discussed in the thread.
所有的问题都只保留一周的开启状态供重点关注解决，超时将关闭，但仍然可以长期继续跟帖讨论。

Please provide a detailed description of the issue and, if necessary, attach a screenshot.
详细描述问题，并在必要时附上屏幕截图。

进入markrom模式，线刷上述包到贝壳云中，第一次上电，无法连接网络，网口灯不亮，路由器端和贝壳云端都不亮，断电后再上电，通过192.168.1.1可以正常进入系统，请问这是正常的吗？

kelvin-ye, 2025-02-10:
不过断电又重启后，网络无法连接，多次断电重启依然无法连接网络。连上显示器如题所示：
[image]

kelvin-ye, 2025-02-10:
又重新烧录，第一次开机，如图，等了接近100秒，才能连上网络，跟之前的现象又有些区别。[image]

进入界面后，用命令重启几次都能正常再连上网络。但一次断电后，就无法正常启动了，提示Failed to connect ubus，如图：[image]

ophub, 2025-02-10:
可能是内核版本的原因

kelvin-ye, 2025-02-10:
> 可能是内核版本的原因
谢谢你的回复，请问要如何解决这个问题，是否需要定位到底是哪一版内核开始出问题的

ophub, 2025-02-11:
你使用的是5.15.y系列的，你可以在构建固件的时候选择这个内核进行构建。

closed 2025-06-02 by ophub