ophub/amlogic-s9xxx-openwrt #696: SW799 板载网卡不稳定

PA733, 2025-02-16:
**Device Information | 设备信息**
 - SOC: RK3399
 - Model: SW799

**OpenWrt Version | 系统版本**
 - Kernel Version: 6.1.128-ophub
 - Release: immortalwrt

**Describe the bug | 问题描述**
[2025.02](https://github.com/ophub/amlogic-s9xxx-openwrt/releases/tag/OpenWrt_immortalwrt_save_2025.02) 版本，release 中的两个包(6.1.128, 6.1.127) 都试过了。开机一段时间后，板载网卡会在短时间内反复重启。USB 网卡 (RTL8153b) 则工作正常。
dmesg 日志：
```
[ 9830.553143] rk_gmac-dwmac fe300000.ethernet eth0: Link is Down
[ 9830.553678] br-lan: port 1(eth0) entered disabled state
[ 9842.842050] rk_gmac-dwmac fe300000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[ 9842.842438] br-lan: port 1(eth0) entered blocking state
[ 9842.842469] br-lan: port 1(eth0) entered forwarding state
[ 9927.834333] rk_gmac-dwmac fe300000.ethernet eth0: Link is Down
[ 9927.834864] br-lan: port 1(eth0) entered disabled state
[ 9930.907747] rk_gmac-dwmac fe300000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
```

PA733, 2025-02-19:
6.1.124 好像没这个问题，路由器目前已稳定运行 6h

ophub, 2025-02-20:
6.6.y问题多点，以后主要使用的内核是6.1.y和6.12.y

PA733, 2025-02-20:
> 6.6.y问题多点，以后主要使用的内核是6.1.y和6.12.y

感谢跟进，我会测试一下 6.12.y 的工作情况。
有什么办法能进一步定位问题吗？看样子是在 6.1.124 - 6.1.127 之间的某些修改出了问题。

PA733, 2025-02-20:
https://github.com/PA733/amlogic-s9xxx-openwrt/actions/runs/13427340026

6.12 好像过不了编译