ophub/amlogic-s9xxx-openwrt #723: 打包时能否指定已编译好的内核

RabbitYilia, 2025-04-08:
**Device Information | 设备信息**
 - SOC: RK3588
 - Model orange pi 5 plus

**OpenWrt Version | 系统版本**
 - Kernel Version: 6.1.99
 - Release: openwrt
**Describe the bug | 问题描述**
想用RK3588专用内核6.1.99 armbian/linux-rockchip@rk-6.1-rkr5 内核，已用amlogic-s9xxx-armbian$ sudo ./recompile -r armbian/linux-rockchip@rk-6.1-rkr5 -k 6.1.y编译出内核文件，用openwrt/openwrt编译出rootfs,请问如何将二者打包？我看了一下代码好像不能指定本地已编译好的内核，只能从仓库下。这个问题该如何解决？顺便求一版6.1内核的固件，现在release只有5.10的

RabbitYilia, 2025-04-14:
./remake -a false -k 6.1.99 -b orangepi-5-plus 重新打包时提示
[ STEPS ] Start making OpenWrt firmware...
 [💔] The [ KERNEL_TAGS ] is invalid: [ rk3588/5.10.y ].

ophub, 2025-04-14:
把本地的那个model.conf里面这个board对应的kernel_tags改成6.1.y