ophub/amlogic-s9xxx-openwrt #739: cp: cannot stat 'bin/targets/*/*/*rootfs.tar.gz': No such file or directory

gdanmac, 2025-04-20:
Run cd imagebuilder
  cd imagebuilder
  [[ -d "output" ]] || mkdir output
  cp -f bin/targets/*/*/*rootfs.tar.gz output/
  cp -f .config output/config
  rm -rf $(ls . | grep -v "^output$" | xargs) 2>/dev/null
  
  # Generate sha256sum file
  cd output/
  for file in *; do [[ ! -d "${file}" ]] && sha256sum "${file}" >"${file}.sha"; done
  
  [[ -d "/builder/openwrt" ]] || mkdir -p /builder/openwrt
  ln -sf /builder/openwrt /home/runner/work/_actions/ophub/amlogic-s9xxx-openwrt/main/openwrt
  df -hT ${PWD}
  echo "status=success" >> ${GITHUB_OUTPUT}
  shell: /usr/bin/bash -e {0}
  env:
    TZ: America/New_York
    IMAGEBUILDER_SH: config/imagebuilder/imagebuilder.sh
    RELEASES_TAG: immortalwrt_24.10.0
cp: cannot stat 'bin/targets/*/*/*rootfs.tar.gz': No such file or directory
Error: Process completed with exit code 1.

gdanmac, 2025-04-20:
打包固件出现Run cd imagebuilder
cp: cannot stat 'bin/targets/*/*/*rootfs.tar.gz': No such file or directory
Error: Process completed with exit code 1. 错误。。请问在哪里可以添加24.10 稳定版？

ophub, 2025-05-19:
你的日志显示你编译失败了，没有生成需要的rootfs.tar.gz文件
需要添加其他分支，在config目录里添加对应的目录和config文件，actions选项里添加对应的选择即可。

closed 2025-06-02 by ophub