ophub/amlogic-s9xxx-openwrt #744: 大佬，编译时只看到了station m1型号，没有看到station m2型号，请问需要如何下载呢？

Kwonelee, 2025-05-11:


Kwonelee, 2025-05-20:
补充主板正反照片：

[image]
[image]

Kwonelee, 2025-05-21:
又会找到了一个dts，请大佬有空可看看！
[rk3566-roc-pc.zip](https://github.com/user-attachments/files/20359277/rk3566-roc-pc.zip)

Kwonelee, 2025-06-03:
群友提供了该设备的u-boot和dts文件，麻烦O大有空看看，是否有帮助！！！感谢

[Station M2-uboot&dts.zip](https://github.com/user-attachments/files/20561369/Station.M2-uboot.dts.zip)

ophub, 2025-06-06:
我测试编译好了，你到releases里下载文件名包含station-m2的固件试试。

Kwonelee, 2025-06-06:
> 我测试编译好了，你到releases里下载文件名包含station-m2的固件试试。

您好！我刷的openwrt_immortalwrt_rockchip_station-m2_k6.12.32_2025.06.06.img这个固件，没能正常从TF卡启动openwrt系统，设备依旧走的eMMC原TV固件。本想着接HDMI看日志，但设备直接就进入TV系统了

ophub, 2025-06-06:
https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_immortalwrt_save_2025.06/openwrt_immortalwrt_rockchip_station-m2_k6.12.32_2025.06.06.img.gz

我改了下打包方法，重新下载这个，解压成img，写卡测试。

Kwonelee, 2025-06-06:
> https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_immortalwrt_save_2025.06/openwrt_immortalwrt_rockchip_station-m2_k6.12.32_2025.06.06.img.gz
> 
> 我改了下打包方法，重新下载这个，解压成img，写卡测试。

感谢大佬，目前TF卡启动正常，系统顺利进入，WIFI可用！
附上启动日志，供您查看！

[系统日志.txt](https://github.com/user-attachments/files/20625255/default.txt)
[内核日志.txt](https://github.com/user-attachments/files/20625285/default.txt)

ophub, 2025-06-06:
好的，感谢反馈，这是第一个碰到需要全量写入的u-boot.bin镜像。我再看看上游文档这类u-boot的差别。打包脚本我已经添加了支持。

openwrt系统启动后可以在浏览器里敲IP进入： 192.168.1.1 默认账号root,默认密码password
在显示器里默认看不到banner欢迎界面，要接键盘回车才能看到。你只要浏览器访问正常就说明可以了，使用openwrt一般也用不到显示器和键盘，直接浏览器访问管理即可。

haizong1, 2026-01-08:
能支持h96max?

jenley, 2026-01-12:
请求添加station P2 的支持,我的设备也是这家的。station P2 请求支持