ophub/amlogic-s9xxx-openwrt #748: 请求添加33a6x的支持

loadnl, 2025-05-20:
不允许上传.dbt文件，压缩了一下，如果不行的话 麻烦大佬看看下面链接里面有

[33a66-Android7.zip](https://github.com/user-attachments/files/20320586/33a66-Android7.zip)

https://files.kos.org.cn/rockchip/SV-33A6X/

ophub, 2025-05-20:
给个设备的详细介绍，什么配置比如几个网卡，是千兆还是百兆、什么芯片、存储、内存等。把设备外观图片发一下，方便的话把主板正反面照片也发一下。

loadnl, 2025-05-20:
[image]

[image]

采用Rockchip RK3399处理器
板载2G内存, 16G eMMC存储
4*USB, 
1*千兆网口, RJ45
1*HDMI, 
1*Mini-PCIe
3.5寸, DC 12V供电
麻烦大佬了 不知道这些信息够不够

ophub, 2025-05-20:
根据你提供的资料我刚添加了这个设备，你到releases里最新的5月固件列表里搜索sv-33a6x下载测试下。

loadnl, 2025-05-20:
> 根据你提供的资料我刚添加了这个设备，你到releases里最新的5月固件列表里搜索sv-33a6x下载测试下。

可以哈 多谢了， 就是固件不能识别wifi，没有无线功能，我自己有装无线网卡

hunter24099, 2025-05-20:
> > 根据你提供的资料我刚添加了这个设备，你到releases里最新的5月固件列表里搜索sv-33a6x下载测试下。
> 
> 可以哈 多谢了， 就是固件不能识别wifi，没有无线功能，我自己有装无线网卡

你好，请麻烦测试两项，一是apt upgrade之后能不能启动，二是dmesg里面有没有报错

ophub, 2025-05-20:
你用的什么wifi芯片，得看用的内核源码里是否有这个驱动。我选的是6.1.y主线内核。

你可以从5.4——-6.12看一下哪个源码有你的无线芯片的驱动，有的话在config里启用编译内核即可

loadnl, 2025-05-22:
> 你用的什么wifi芯片，得看用的内核源码里是否有这个驱动。我选的是6.1.y主线内核。
> 
> 你可以从5.4——-6.12看一下哪个源码有你的无线芯片的驱动，有的话在config里启用编译内核即可

嗯嗯 测试可以了 麻烦老大了哈 多谢了

hunter24099, 2025-05-24:
> > 你用的什么wifi芯片，得看用的内核源码里是否有这个驱动。我选的是6.1.y主线内核。
> > 你可以从5.4——-6.12看一下哪个源码有你的无线芯片的驱动，有的话在config里启用编译内核即可
> 
> 嗯嗯 测试可以了 麻烦老大了哈 多谢了

请问出现文件系统只读的情况吗？刷了bookworm和noble都是这个问题

loadnl, 2025-05-24:
大佬 我拷了你的库 然后在自己库里面 用action编译        
       uses: ophub/amlogic-s9xxx-openwrt@main   ——> 改成了 loadnl/amlogic-s9xxx-openwrt@main  
        with:
          openwrt_path: openwrt/output/*rootfs.tar.gz
          openwrt_board: ${{ inputs.openwrt_board }}
          openwrt_kernel: ${{ inputs.openwrt_kernel }}
          auto_kernel: ${{ inputs.auto_kernel }}
主要目的是修改 openwrt_kernel 因为我在你回答其他人的信息里面看到要修改对应的 KERNEL_TAGS    我把默认的stable/6.1.y 改成了目标stable/5.15.y ，编译过程中提示/home/runner/work/_actions/ophub/amlogic-s9xxx-openwrt/main/make-openwrt/kernel/stable/5.10.y.tar.gz: Cannot open: No such file or directory  后来我就修改配置指定了固定版本号 编译通过，但是安装后运行不了，hdmi也没画面 ，能麻烦你帮忙看看原因吗？这是编译成功的日志：https://github.com/loadnl/amlogic-s9xxx-openwrt/actions/runs/15222847109/job/42821033907