ophub/amlogic-s9xxx-openwrt #754: s905x2 takes wrong dtb in openwrt-install-amlogic command

drsamir21, 2025-05-25:
My device H96maxX2 (4GB , gigabit ethernet)
DTB meson-g12a-x96-max-rmii.dtb

I can change dtb file name in uEnv.txt file when I use it from SDCARD and my device works good.
When I use openwrt-install-amlogic command in terminal there's is no option to choose my device but compulsorily I chose 301 that is proper in 4GB RAM and gigabit ethernet support but it took meson-g12a-x96-max.dtb which is wrong. So give proper option to choose meson-g12a-x96-max-rmii.dtb

drsamir21, 2025-05-25:
[image]

ophub, 2025-05-25:
choose `0`
Input DTB Name

drsamir21, 2025-05-26:
I tried 0 , but then  I need input all values it asks but anyway it installed to emmc which is a bit annoying.

Then I tried to use openwrt-update-amlogic command to update again to check but in this command it took boot partition data from image file which is again incorrect. To solve this I had to edit image file and edited there which is again annoying.

[image]