ophub/amlogic-s9xxx-openwrt #759: 一直卡在 正在加载视图 ...78秒

ihuanx, 2025-05-29:
openwrt_immortalwrt_amlogic_s905d_k6.12.28_2025.05.19.img

ihuanx, 2025-05-29:
哪个是 immortalwrt 还是opkg包管理器的版本?

ihuanx, 2025-06-05:
OpenWrt_armv8_save_2025.06
OpenWrt_lede_save_2025.06 
这两个不知道为什么U盘启动不了 

OpenWrt_immortalwrt_save_2025.06 
这个卡视图

OpenWrt_official_save_2025.06 
这个好像不卡视图 

测试平台 s905d _N1 

好想要一个24.10.1版本的 包管理器还是opkg的版本.    快照版本没有 mosdns 5.3.1.apk

ImmortalWrt 最新的 mosdns-5.3.3-r1.apk  N1用不了 = =

[luci-app-mosdns](https://github.com/sbwml/luci-app-mosdns) 又只有IPK文件

closed 2025-06-05 by ihuanx

ophub, 2025-06-06:
OpenWrt_armv8_save_2025.06
OpenWrt_lede_save_2025.06
这两个都是lede的源，armv8是使用f大的打包脚本，下面是我的打包脚本。理论上结果相同。

OpenWrt_immortalwrt_save_2025.06
OpenWrt_official_save_2025.06
这2个理论上结果相同，都使用了JavaScript脚本加载页面。卡加载你得仔细看卡什么地方了，浏览器F12打开开发者模式，观察网络加载那个栏目，看看哪个页面或者文件加载时间长，找到卡哪里了才好解决。卡一般是调用查询某个数据反应时间过长或者报错导致的。

fansuregrin1, 2026-10-30:
大佬您好。我从U盘启动openwrt，用的 [openwrt_lede_amlogic_s905d_k6.18.56_2026.10.15.img.gz](https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_lede_master_save_2026.10/openwrt_lede_amlogic_s905d_k6.18.56_2026.10.15.img.gz)，浏览器访问后台也是卡“加载视图”。

[image]