ophub/amlogic-s9xxx-openwrt #760: OpenWrt和immortalwrt刷完后，在哪更改软件包源？

8mayongde2, 2025-06-03:
上次刷完这两个固件装passwall失败，命令符装.run包提示没软件包源。
我想更换成kwrt软件包源，请问 在哪里更改？
src/gz kwrt_core https://dl.openwrt.ai/releases/24.10/targets/armsr/armv8/6.6.89
src/gz kwrt_base https://dl.openwrt.ai/releases/24.10/packages/aarch64_generic/base
src/gz kwrt_packages https://dl.openwrt.ai/releases/24.10/packages/aarch64_generic/packages
src/gz kwrt_luci https://dl.openwrt.ai/releases/24.10/packages/aarch64_generic/luci
src/gz kwrt_routing https://dl.openwrt.ai/releases/24.10/packages/aarch64_generic/routing
src/gz kwrt_kiddin9 https://dl.openwrt.ai/releases/24.10/packages/aarch64_generic/kiddin9

设备是hk1box

ophub, 2025-06-03:
系统---软件包--配置

8mayongde2, 2025-06-06:
> 系统---软件包--配置

昨晚用U盘适用了两个包；
openwrt_official_amlogic_s905x3_k6.12.31_2025.06.01固件包换不了软件包源，在软件包页上传ipk安装报错。
openwrt_s905x3_multi_R25.05.05_k6.12.31-ophub固件包软件包页上传ipk安装报错，因时间问题没测其它。
ipk文件是openclash，ipv6是通的。

closed 2025-08-01 by ophub