ophub/amlogic-s9xxx-openwrt #765: Clock Speed & Total Available Memory (RAM)

bulzed, 2025-06-14:
**Device Information | 设备信息**
 - SOC: s922x
 - Model: Hardkernel ODROID-N2Plus

**OpenWrt Version | 系统版本**
 - Kernel Version: 6.6.93-ophub
 - Release: ImmortalWrt 24.10.1 r33048-cc720ea55a71

**Describe the bug | 问题描述**
1. Why I can't get my real total available RAM, which is should be 4GB instead of only 2GB?
2. Why the clock speed doesn't match between what it says in LuCI (1000MHz) and terminal (1800MHz)?

Please provide a detailed description of the issue and, if necessary, attach a screenshot.
详细描述问题，并在必要时附上屏幕截图。

[image]

[image]

bulzed, 2025-06-14:
do you mean this file (/boot/u-boot-s905x2-s922.bin) that you talk about?
and please tell me where's the link for repository to check, so I can try it to test.
thanks

ophub, 2025-06-14:
https://github.com/ophub/amlogic-s9xxx-armbian/tree/main/documents#1212-error-in-memory-size-recognition

bulzed, 2025-06-18:
I have try the u-boot file from https://github.com/ophub/u-boot/blob/main/u-boot/amlogic/overload/u-boot-odroid-n2.bin still nothing work, sir..

closed 2025-06-18 by bulzed