ophub/amlogic-s9xxx-openwrt #771: ophub大佬能将农商云nsy-g68-plus路由器加入支持？

faq110, 2025-06-19:
ophub大佬能将农商云nsy-g68-plus路由器加入支持吗？
设备配置：rk3568 cpu，4G ddr4，交换机芯片RTL8367S，WiFi芯片MT7916D，1个USB3.0，5个千兆网口，1条M2插槽，EMMC 8G或更大

已有的编译https://github.com/xiaomeng9597/OpenWrt-NSY/releases

机器图片如下：

[image]

[image]

[image]

ophub, 2025-06-19:
https://github.com/ophub/amlogic-s9xxx-openwrt/releases/download/OpenWrt_official_save_2025.06/openwrt_official_rockchip_nsy-g68-plus_k6.1.141_2025.06.15.img.gz

下载测试下能不能使用

yuyu913gzs, 2025-07-28:
https://github.com/zhoufuli/immortalwrt-rk356x这个里有驱动和dts文件

ophub, 2025-12-03:
添加好了，你到releases里下载测试下：

https://github.com/ophub/amlogic-s9xxx-openwrt/releases

linai081795, 2025-12-04:
这个 RTL8367S 交换机网卡 无法驱动 能正常启动并且能进入系统 我测试的是 armbian 我的机型是 BDY-G18-Pro 能使用 TTL 调试、和使用USB 网卡

[image]

linai081795, 2025-12-04:
我看到有openwrt 和 lede 的驱动 主线版本的RTL8637S 我没找到

[rtl8367b-fix-gmac.tar.gz](https://github.com/user-attachments/files/23933852/rtl8367b-fix-gmac.tar.gz)
[rtl8367b-fix-gmac-lede.tar.gz](https://github.com/user-attachments/files/23933851/rtl8367b-fix-gmac-lede.tar.gz)

[rtl8367b.tar.gz](https://github.com/user-attachments/files/23933861/rtl8367b.tar.gz)
[这是对方仓库](https://github.com/xiaomeng9597/files/releases)

ophub, 2025-12-04:
@linai081795 
针对你的设备bdy-g18-pro，我转码开了一贴进行这个设备的持续讨论跟踪：
https://github.com/ophub/amlogic-s9xxx-openwrt/issues/847

移步到这个帖子下集中讨论你的设备的问题，我刚才在仓库里添加了这个设备，昨天编译了u-boot和dtb，但是没找到有人发帖就没添加，你有设备咱们就在这个帖子下跟踪，争取把它适配好。

HanRuo, 2025-12-09:
大佬们好，我现在使用的设备是G68这款设备，使用的是周佬的仓库（[https://github.com/zhoufuli/immortalwrt-rk356x]）所编译的immortalwrt兼istore的固件，使用下来精简过多，虽然额外编译了core/cake等相关依赖，但是实际使用下来QOS支持不是很好，另外想要加入硬件解码，也一直不得其法；

https://github.com/xiaomeng9597/ImmortalWrt-NSY  xiaomeng仓库在周佬仓库基础上的编译，可以将SQM等相关内容编译进去，但是luci-app-mtk.ipk 却无法被正常编译进去，导致网络配置下无WIFI配置选项。

如大佬有时间和精力，希望可以补足设备硬解及其他短板，感谢