ophub/amlogic-s9xxx-openwrt #803: luci-app-amlogic新版提示找不到dtb文件，无法写入emmc，老版本可以

teddy780102, 2025-08-02:
luci-app-amlogic新版提示找不到dtb文件，无法写入emmc，老版本可以

ophub, 2025-08-02:
什么设备，发提示信息的截图，发插件版本截图

teddy780102, 2025-08-02:
3.1.261不行，3.1.256正常写入emmc

ophub, 2025-08-02:
我测试使用正常，晶晨宝盒最新版3.1.261版本，测试n1写入emmc没问题

[image]
[image]
[image]
[image]

closed 2025-08-02 by teddy780102